iNavFlight/inav #3670: Feature Request : Autoland in a straight line

PaulBizard, 2018-07-28:
_Is this a new feature request?_ YES

Hi everybody, just chiming in. I just bought my first INAV flight controller and was wondering if anybody considered the "Autoland in a straight line Mode".

See the drawing attached below.
[image]

The user would just specify the impact point coordinates, the direction of the landing, the initial altitude, the glide slope and the radius of the two Heading Alignment Cylinders.

Paul

hali9, 2018-07-30:
Yes. https://github.com/iNavFlight/inav/issues/1814

PaulBizard, 2018-07-31:
Thanks hali9, I've had a look at the previous idea.

My way of doing is inspired by the orbiter.

1/ Lateral guidance: tangent the nearest cylinder (if inside then first head straight until outside of cylinder). Longitudinal guidance: go to specified altitude

2/ if aligned with the runway and if at the good altitude then start glide to the runway.

closed 2018-10-13 by stale[bot]

teckel12, 2018-11-19:
@davidngrc I believe #1814 is basically the same request which is still open. So there's no reason to dilute it by having two issues open.