iNavFlight/inav #3671: vtx_low_power_disarm no longer available ?

wilco1967, 2018-07-28:
INAV/MATEKF722 2.0.0 Jul 28 2018 / 13:34:26 (cf4048d24)
custom compiled development firmware

I've been using the automatic VTX power reduction before, and it really was a nice feature...

But I just noticed, it's no longer available in the latest development version.... 
I always compare dump files between versions, and the following settings have disappeared.
set vtx_band = 3
set vtx_channel = 1
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5705
set vtx_pit_mode_freq = 0

A dump from today's version doesn't have those parameters anymore....

Seems to have disappeared somewhere between 22-7 and 28-7 (today)
Don't see any commits mentioning this .

Is this accidental, or on purpose ?

I really liked the feature. Hope it comes back

shellixyz, 2018-07-28:
This is involuntary. Probably related to #3656. Must be an issue with the updated condition for the VTX settings. I'm going to take a look. In the meantime you can just comment the line 1667 in the `fc/settings.yaml` file.

`#condition: defined(VTX_COMMON) || (defined(USE_CMS) && defined(USE_VTX_SMARTAUDIO))`

wilco1967, 2018-07-28:
yeah, that seems to have worked..... they are back again......

Thanks a lot !!!!

P-I-Engineer, 2018-07-28:
Mine is working in rc4, smart audio, low power on first on

wilco1967, 2018-07-28:
yes, in RC4 it still worked, but sometime after RC4 in the development branch, it stopped working....

guess it will be corrected in the next RC or final 2.0.0...
Commenting out that line 1667 makes it working again.

P-I-Engineer, 2018-07-28:
sorry, didn't understand, please ignore

wilco1967, 2018-07-28:
never mind...... I often don't understand this complex stuff as well..... just doing my best..

closed 2018-08-19 by wilco1967