iNavFlight/inav #3672: Softserial not working on Omnibus f4 pro v3 after 2.0rc4 update

bokie1, 2018-07-29:
I tries updating my omnibus f4 pro v3 to 2.0rc4 from 1.9.1 but telemetry is not working anymore. I have frsky rx on soft serial 1 port. Didn't change anything on the wing, just the fw.
Already tries setting up from scratch, didn't work. Also tried copying diff, didn't work also. 
I have another wing but with omnibus f4 v3, everything still works when I upgraded to 2.0rc4. Only issue I got on omnibus f4v3 after update is the ps, but that's on another thread already.
Any suggestions.

teckel12, 2018-07-30:
@bokie1 What RX do you have? On one quad with an Omnibus F4 Pro v3 that uses an XSR RX, I needed to `set smartport_uart_unidir = ON` for telemetry to function. With another quad with the same FC and very similar hardware (other than the RX) I could leave this as `OFF` and it would work.

I opened a ticket for this but haven't had time to investigate further.

bokie1, 2018-07-30:
@teckel12 I have r9 slim.

teckel12, 2018-07-30:
@bokie1 A `diff` would be helpful.

bokie1, 2018-08-04:
I tried the same fw on another omnibus f4 pro v3 board and the same result. So it's not the board. Another thing I tested was the same fw but on a clone omnibus f4 pro v2, lo and behold, everything works. 
I can conclude now that it's only affecting the omnibus f4 pro v3 boards.
I hope someone can take a look at this issue.

teckel12, 2018-08-04:
@bokie1 I have two omnibus f4 pro v3 and have no trouble with SmartPort telemetry. The only issue I have is with my XMR Rx I need to set

bokie1, 2018-08-04:
@teckel12, are you using r9 slim/+/mini rx?

teckel12, 2018-08-04:
XSR-M and X4R-SB

bokie1, 2018-08-04:
Oh ok. Maybe try to replicate it once you have an r9 series rx. It also works with x series rx for me.