iNavFlight/inav #3677: Firmware Request for Speedy Bee F4 AIO Flight Controller

datamann, 2018-07-30:
### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_

https://www.speedybee.com/speedy-bee-f4-aio-flight-controller/

_Version of INAV used goes here_ Latest

### Behavior
* _Describe the problem_ No support. BF is supported.

_Is this a new feature request?_ Probably

stronnag, 2018-07-30:
You mean like https://github.com/iNavFlight/inav/pull/3653 ?

datamann, 2018-07-30:
Exactly! Perfect :-)

shellixyz, 2018-07-30:
Starting from 2.0 RC5

datamann, 2018-08-07:
Thanks for releasing INAV 2.0.0 RC5 but it seems like firmware for Speedy Bee F4 AIO FC did not make it for this release?

datamann, 2018-08-07:
When I look here:
https://github.com/iNavFlight/inav/releases
I cannot find "SPEEDYBEEF4" but if I download sourcecode:
https://github.com/iNavFlight/inav/archive/2.0.0-RC5.zip

I do find it:
\inav-2.0.0-RC5\src\main\target\SPEEDYBEEF4

So I guess I can compile it myself even though I have never done it before...

giacomo892, 2018-08-07:
SPEEDYBEEF4 is not yet in the official release make process. @digitalentity should we add it into make release?

digitalentity, 2018-08-07:
@giacomo892 will update

closed 2018-08-07 by digitalentity