iNavFlight/inav #3679: 2.0.0 RC4 RTH problem

lo4wai, 2018-07-31:
I flew my 7 inch quad with inav 2.0 RC4 last weekend.
[Video footage](https://youtu.be/_9vJVLQTY6M)

My vtx (VT5804m V2) lost signal suddenly in mid flight.  It seems that the TRAMP communication was lost and vtx fall back to it's "default" channel (I checked the solder joint is fine, will test with another vtx to see if it is hardware problem).  Therefore I enable RTH to bring the quad back

In 1:15 of the video, after the quad climb to desire altitude, it flew at full speed with nearly 100% throttle and could not keep the altitude (I lately discovered that the nav_auto_speed is accidentally set to 10000).  Once I spot the quad, I switched to POSHOLD mode and tried to bring the quad back manually.  However I am poor at LOS and forgot the quad is heading home.  I pitch back and the quad fly away.

I tried to enable RTH again at 1:58, however the quad didn't climb and fell slowly into ground instead.

Here is my blackbox log
[RTH not climb.TXT](https://github.com/iNavFlight/inav/files/2245641/RTH.not.climb.TXT)

I also found that sometimes RTH overshoot or spin above home (around RTH at 2:56 and spin at 3:07 of the blackbox below)
[RTH spining.TXT](https://github.com/iNavFlight/inav/files/2245644/RTH.spining.TXT)

Here is my config below
```# diff

# version
# INAV/MATEKF405 2.0.0 Jul 24 2018 / 12:47:18 (f2cf315)

# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# feature
feature GPS
feature LED_STRIP
feature BLACKBOX
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper
beeper -ON_USB

# map

# serial
serial 0 2 115200 57600 0 115200
serial 4 4096 115200 38400 0 115200

# led
led 0 11,15::C:5
led 1 10,15::C:5
led 2 9,15::CBW:1
led 3 8,15::CBW:1
led 4 7,15::CBW:1
led 5 6,15::CBW:1
led 6 5,15::C:2
led 7 4,15::C:2

# color

# mode_color

# aux
aux 0 0 0 1100 2100
aux 1 1 0 1250 1350
aux 2 9 0 1400 1500
aux 3 8 0 1550 1650
aux 4 11 1 1300 2100
aux 5 18 0 1900 2100

# adjrange

# rxrange

# osd_layout
osd_layout 0 0 1 3 V
osd_layout 0 1 24 1 V
osd_layout 0 2 0 0 V
osd_layout 0 7 12 11 V
osd_layout 0 9 1 6 H
osd_layout 0 10 23 7 V
osd_layout 0 11 24 3 V
osd_layout 0 12 24 4 V
osd_layout 0 13 3 9 V
osd_layout 0 14 1 0 V
osd_layout 0 15 2 10 V
osd_layout 0 20 6 0 V
osd_layout 0 21 17 0 V
osd_layout 0 22 12 10 V
osd_layout 0 23 14 10 V
osd_layout 0 25 22 4 V
osd_layout 0 30 1 12 V
osd_layout 0 31 1 1 V
osd_layout 0 32 24 2 V
osd_layout 0 33 1 6 V
osd_layout 0 34 10 2 V
osd_layout 0 41 25 10 V
osd_layout 0 42 25 9 V

# master
set gyro_sync = ON
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 80
set gyro_notch1_cutoff = 100
set gyro_stage2_lowpass_hz = 150
set acc_hardware = MPU6000
set acczero_x = 76
set acczero_y = -38
set acczero_z = -159
set accgain_x = 4086
set accgain_y = 4111
set accgain_z = 4039
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set mag_declination = -250
set magzero_x = -189
set magzero_y = -148
set magzero_z = 267
set baro_hardware = BMP280
set pitot_hardware = NONE
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set min_throttle = 1035
set max_throttle = 2000
set motor_pwm_rate = 2000
set motor_pwm_protocol = ONESHOT125
set failsafe_procedure = RTH
set model_preview_type = 3
set small_angle = 180
set gps_sbas_mode = MSAS
set nav_position_timeout = 1
set nav_wp_radius = 5
set nav_wp_safe_distance = 100
set nav_auto_speed = 10000
set nav_auto_climb_rate = 300
set nav_manual_climb_rate = 500
set nav_land_slowdown_minalt = 200
set nav_land_slowdown_maxalt = 500
set nav_emerg_landing_speed = 2000
set nav_min_rth_distance = 0
set nav_rth_abort_threshold = 500
set nav_max_terrain_follow_alt = 50000
set nav_rth_altitude = 2000
set nav_rth_home_altitude = 500
set nav_mc_bank_angle = 40
set nav_mc_hover_thr = 1330
set nav_mc_auto_disarm_delay = 1330
set nav_fw_cruise_thr = 15
set nav_fw_min_thr = 0
set nav_fw_max_thr = 1200
set nav_fw_bank_angle = 208
set nav_fw_climb_angle = 7
set nav_fw_dive_angle = 20
set nav_fw_pitch2thr = 164
set nav_fw_loiter_radius = 10
set nav_fw_cruise_speed = 1400
set nav_fw_land_dive_angle = -120
set nav_fw_launch_velocity = 2
set nav_fw_launch_accel = 300
set nav_fw_launch_max_angle = 0
set nav_fw_launch_detect_time = 1863
set nav_fw_launch_thr = 1000
set nav_fw_launch_idle_thr = 40
set nav_fw_launch_motor_delay = 1700
set nav_fw_launch_spinup_time = 500
set nav_fw_launch_min_time = 100
set nav_fw_launch_timeout = 0
set nav_fw_launch_max_altitude = 5000
set nav_fw_launch_climb_angle = 0
set nav_fw_cruise_yaw_rate = 18
set nav_fw_allow_manual_thr_increase = VALUE 45 OUT OF RANGE FOR nav_fw_allow_manual_thr_increase

set osd_video_system = NTSC
set osd_rssi_alarm = 40
set osd_left_sidebar_scroll = GROUND_SPEED
set osd_right_sidebar_scroll = ALTITUDE
set osd_sidebar_scroll_arrows = ON
set debug_mode = GYRO
set tz_offset = 480
set vtx_band = 5
set vtx_channel = 4
set vtx_power = 2
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5769

# profile
profile 1

set mc_p_pitch = 63
set mc_i_pitch = 50
set mc_d_pitch = 33
set mc_p_roll = 48
set mc_i_roll = 45
set mc_d_roll = 32
set mc_p_yaw = 70
set max_angle_inclination_rll = 400
set max_angle_inclination_pit = 400
set dterm_lpf_hz = 80
set dterm_notch_cutoff = 100
set thr_mid = 33
set thr_expo = 70
set rc_expo = 80
set rc_yaw_expo = 80
set roll_rate = 108
set pitch_rate = 108
set yaw_rate = 98

# battery_profile
battery_profile 1
```

kn8alpha, 2018-08-01:
set nav_rth_abort_threshold = 500 which is 5 meters
If quad overshoots set position during RTH by 5 meters it quits RTH and does emergency landing.. ?

lo4wai, 2018-08-01:
Thanks for spotting that!
I am not sure that I have set this variable before, I will try to reset to defaults and set the quad up again.

fiam, 2018-08-01:
`set nav_fw_allow_manual_thr_increase = VALUE 45 OUT OF RANGE FOR nav_fw_allow_manual_thr_increase`

This means your settings are corrupted, you probably upgraded between RCs without doing a full erase. I'd suggest you take a `diff` do a full erase and restore the settings one by one carefully.

lo4wai, 2018-08-01:
I probably forgot that!!! Thanks very much!!!

closed 2018-08-01 by lo4wai