iNavFlight/inav #3680: v2.0.0RC4 R9 Mini + Omnibus f4 pro clone FPort on Uart1 (no telemetry)

davidngrc, 2018-07-31:
ok, I found out the FPort is working on Omnibus F4 Pro v2 + softserial on CH5 and CH6 pad + R9 mini (without the inverter) + iNav 2.0.0RC4    (check here https://github.com/iNavFlight/inav/issues/3344#issuecomment-409256384 )

But when I try to do the same thing on UART 1, I got stick control, but no telemetry.

Wiring 
Omnibus F4 Pro CH5 pad (RX) -> 1k resister -> CH6 pad (TX)
Omnibus F4 Pro CH6 pad (TX) -> R9 mini smart port (do not need to connect the sbus wire)
CH5 CH6 pad are on https://github.com/iNavFlight/inav/issues/1346#issuecomment-335654194

in INav Configurator 2.0.0 RC3
1)  in "configuration tab", enable soft serial
2) in "port tab", enable "serial rx" on uart1
3)  in "configuration tab", receiver mode set to "serial based receiver, and provider set to fport (rather than sbus)
4)  in "CLI tab", set serialrx_inverted = OFF or ON ( doesn't matter ）
                    set serialrx_halfduplex = OFF
5) now on "Receiver tab", you should able to see the channel value change while you move the x9d stick.
6) on x9d, delete all old telemetry sensor in "telemetry screen", then discover new.
7) no new sensor found, bug?

teckel12, 2018-07-31:
@davidngrc I seem to remember a pull request for a change to the inverted receiver and/or telemetry.  I believe it was included in RC4 but not 100% sure.

Also, how do you have it wired from the R9 mini to the FC?

davidngrc, 2018-07-31:
@teckel12 as mention above, Omnibus F4 Pro sbus (uart1) -> R9 mini smart port
R9 mini 5v from BEC, not from FC.

BTW, I could not get the FPort working on matek F405 Wing on iNav 2.0.0RC4
if you can, could you share the dump?

teckel12, 2018-07-31:
@davidngrc One wire? To what pin in the RX and what pin on the FC?

davidngrc, 2018-07-31:
@teckel12 in Omnibus f4 pro, I link it to the sbus pin, I believe it is a tx pin
in matek F405 Wing, I try uart 2, 3, softsearil1, all tx pin without success

bokie1, 2018-08-01:
R9 mini doesn't have an uninverted s.port. Therefore, it needs an inverter for f.port to work, unless it has changed.

bokie1, 2018-08-01:
I think this issue is related to my s.port telemetry not working.

teckel12, 2018-08-01:
@davidngrc Why are you using an inverter? FPort isnt inverted. It's designed to run from an uninverted port on the FC to uninverted pads on the RX.

davidngrc, 2018-08-01:
@teckel12  because the sbus pin is inverted in F4 board?

davidngrc, 2018-08-02:
@teckel12  wait a second....
if it is a RX protocol, that means FPort not suitable for long range flight?

because I don't think the R9mini can output 1W of power?
while we use sbus, sometimes we get telemetry lost (RX side), but still flyable, this is due to the awesome R9M 1W power (TX side)?

anyone test the FPort range vs the SBUS range?
same or not?

teckel12, 2018-08-02:
@davidngrc Use the other port for the same UART that isn't inverted. There's another connection for the serial comm port that's not inverted. No need for this double inversion nonsense. That's what FPort is designed to resolve.

teckel12, 2018-08-02:
@davidngrc The range should be about the same as S.Bus. Same hardware, just the protocol changes.

davidngrc, 2018-08-03:
@teckel12  Just try FPort on Omnibus f4 pro clone 

TX6 pin (without sbus invertor),
try all the combination of serialrx_inverted  and  serialrx_halfduplex  in CLI, 
no stick control and telemetry.

RX6 pin (without sbus invertor),,
same step above, same no result.

TX6 pin (with sbus invertor, also try switch sbus invertor IN and OUT),
same step above, same no result.

RX6 pin (with sbus invertor, also try switch sbus invertor IN and OUT),
same step above, same no result.

so FPort is indeed has bugs, not work on tx1 (inverted) or tx6 (non-inverted, with sbus invetor or not).
only port working seems to be the softserial1 (**without** sbus invetor) with CLI setting  serialrx_inverted = ON  and serialrx_halfduplex = ON

teckel12, 2018-08-03:
@davidngrc How are you connecting to your receiver? To the SBUS and S.Port pins (which are inverted) or the uninverted SBUS and S.Port pads as you should be for FPort?

On the F4 you should really be using an uninverted UART and the uninverted pads on your receiver.  

You can't invert the UART signals with software switches on most F4 FC so your making things more complex by even trying that.

The duplex option is for one wire, so again, if your using two wires to two UART pins you're complicating things for no reason.

The problem isn't with FPort, it's with what your trying to do, how you're doing it, and the complexity you're adding for no reason (making it all more confusing than it should be).

So step one, forget the inverters. Step two, use a physical UART on your FC, wire the RX pad on the FC to the UNINVERTED SBUS pad on your receiver. Wire the TX pad on the FC to the UNINVERTED S.Port pad on your receiver. Step three, don't set any inverter settings, just FPort and SmartPort telemetry.

bokie1, 2018-08-03:
@teckel12 I thought Fport is just 1 wire on TX? Why woupd we need another cable to be connected on Sbus/rx pad?
I understand why @davidngrc is using softserial because r9 mini doesn't have an uninverted s.port pad.
For my setup, I only use sbus (rx1) and s.port (tx11/ch6). It used to work on 1.9.1 but stopped working on 2.0. Tested on rc2,3 and 4.

bokie1, 2018-08-03:
@davidngrc, omnibus f4 pro doesn't have an uninverted uart tx. It has an uninverted uart rx on rx1. That is why, either we need to use tx11 or use a hardware inverter to make s.port or f.port work.

davidngrc, 2018-08-03:
@teckel12  my wiring is very simple, one wire from R9 mini SPort to FC UART TX or RX.

like @bokie1 said, R9 mini does not come with an uninverted s.port, so I add a sbus invertor to the existing inverted sport, that means I should get a uninverted sport signal.

I know FPort is suppose to connect to the FC TX pad.
but just in case, I also test on the RX pad as well.

@teckel12  do you have a  Omnibus f4 pro clone?
 if so, could you make FPORT work on Uart1 or 3 or 6?
I have proved that it will work on softserial 1, but my test show it can not work on Uart 1 or 6, I assume 3 will be the same.

davidngrc, 2018-08-03:
@bokie1 I have add the hardware invertor in my test, still don't work.

bokie1, 2018-08-03:
@davidngrc, which inverter are you using? Mine stopped working after 1.7.3.

davidngrc, 2018-08-03:
this one, search "sbus inverter" on google
[image]

bokie1, 2018-08-03:
That's sbus inverter not s.port inverter.

davidngrc, 2018-08-03:
I read from somewhere, it said, it is the same thing.
both invert the signal.

ninja-zx11, 2018-08-03:
I believe it has to be bi directional inverter.Check my thread here:
https://www.rcgroups.com/forums/showthread.php?3112594-F-port-on-R-XSR-via-bi-directional-inverter-not-working-%28Solved%29/page2#post40025930

bokie1, 2018-08-03:
@ninja-zx11 that's the inverter I was using with my xsr-m before. Sadly, it stopped working after 1.7 3.
With regards to the swapping of resistors to lesser values, did it also work for s.port or it was just a fix for f.port?

teckel12, 2018-08-04:
I did get FPort working, using uninverted signals. It probably isn't working for you because your not using a bi-directional inverter. That's why I said for an F4 you really need to go totally uninverted. I'm not going to bother with FPort till I build a F7 quad, not really designed for the F4 if you're using a RX that doesn't have uninverted pads exposed.

davidngrc, 2018-08-04:
thank you all, can't find bi-directional inverter in online shop, seems I had to make one myself.
could someone update the wiki include the useful info here?
thank you again😄

closed 2018-08-04 by davidngrc

teckel12, 2018-08-04:
@davidngrc There's an extensive (I believe BF) wiki on FPort which points out the problems with using FPort on an F4 FC.

davidngrc, 2018-08-04:
@shellixyz @teckel12 
It would be even better include the info in below thread 😅 

https://www.rcgroups.com/forums/showthread.php?3112594-F-port-on-R-XSR-via-bi-directional-inverter-not-working-%28Solved%29

davidngrc, 2018-08-04:
@shellixyz @teckel12  very helpful. thank you

cterraza, 2019-01-11:
According to this page:

https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20Omnibus%20F4.md

The Uart1 and Uart6 has a unidirecctional inverter on the TX pins, so the FPORT cannot work in that UARTS.

I´m already make it work with any issue in the UART3, but this UART is shared with the I2C pins. So, if you have a external compass or another sensor with I2C protocol this can´t work anymore.

The CLI commands that I used on INAV 2.0.0 was:

set serialrx_provider = FPORT
set serialrx_inverted = OFF
set serialrx_halfduplex = ON

Obviously, you need to use the uninverted FORT pin on the R9mm.

[image]