iNavFlight/inav #3681: AKK Ultimate X2 Smart Audio Problem with F35 Lightning FC

haksarion, 2018-07-31:
### Board and Version
Furious FPV F35 Lightning FC Full Options
Inav 2.0 RC 4

### Behavior
When I turn on "TBS Smart Audio" on Port 4 in Ports Tab Smart Audio seems to work first but if I try to change any settings in OSD it does not change anything.
Secondly if I use TBS Smartaudio the new Feature that automatically powers VTX Power low and high has only 4 levels.

Greetings Hakan

lucaslabs, 2018-08-08:
Second this one, also is not working on RC5. Please fix it. 
Thanks!

lucaslabs, 2018-08-16:
hey @stronnag @fiam what info do we need to provide so you can fix this one? 😃

digitalentity, 2018-08-16:
2.0-RC6 SA bugs seem to be have common root cause. Tracked in metabug #3753

closed 2018-08-16 by digitalentity