iNavFlight/inav #3686: My F4-CTR flight controller could not open the barometer and direction compass, both icons are lit red.

chenxianan, 2018-08-01:
Refresh the latest 1.9.1 matekf405 firmware, connect to the GPS and direction compass (beitian BN-880), open the control station to the flight controller to display the direction compass, and the barometer is illuminated red. I have two options to rotate the barometer and direction compass separately in the configuration item. The I2C error in the lower left corner shows the 205.5.200.216 prompt. GPS is useful. After the configuration is correct, the GPS can read the location information. When I don't have a power battery connected, it only lights up red. But I turned off the barometer and direction compass options, I2C shows 000, so I don't understand where it appears. Ask your question, (GPS rx connection controller tx, tx connection rx.sda connection sda, scl connection scl.v5 connection v5, gnd connection gnd)[
[image]
[image]
[image]
[image]

](url)

Tygrys-1, 2018-08-05:
Is I2C properly terminated/does it have pullup resistors?

Tygrys-1, 2018-08-08:
Try connecting the battery. #3688

chenxianan, 2018-08-09:
问题已经解决，是方向罗盘硬件损害更换之后就没有问题了，而且LED不能悍接在LED位置要焊接在s5的位置

closed 2018-08-09 by chenxianan