iNavFlight/inav #3690: Convert blackbox CSV rea

col85, 2018-08-02:
Hi all, I am trying convert the decoded blackbox CSV data into actual conventional sensor readings (e.g. instead of accSmooth values, how can I convert this to actual accelerometer readings?) that we are viewing on Blackbox explorer for my own analysis. Would like to ask if there are any guidelines on how I can do this?

Thanks in advance for the advice!

shellixyz, 2018-08-03:
Apparently value in g = accSmooth / 2048

shellixyz, 2018-08-03:
Looks like it depends on the accelerometer:

```
drivers/accgyro/accgyro_bma280.c:44:    acc->acc_1G = 512 * 8;
drivers/accgyro/accgyro_mpu6050.c:112:        acc->acc_1G = 512 * 4;
drivers/accgyro/accgyro_mpu6050.c:115:        acc->acc_1G = 256 * 4;
drivers/accgyro/accgyro_mma845x.c:98:    acc->acc_1G = 256;
drivers/accgyro/accgyro_mpu6500.c:46:    acc->acc_1G = 512 * 4;
drivers/accgyro/accgyro.h:63:    uint16_t acc_1G;
drivers/accgyro/accgyro_mpu9250.c:46:    acc->acc_1G = 512 * 4;
drivers/accgyro/accgyro_bmi160.c:211:    acc->acc_1G = 4096;
drivers/accgyro/accgyro_adxl345.c:68:    acc->acc_1G = 256; // 3.3V operation
drivers/accgyro/accgyro_mpu6000.c:136:    acc->acc_1G = 512 * 4;
drivers/accgyro/accgyro_lsm303dlhc.c:118:    acc->acc_1G = 512 * 8;
```

shellixyz, 2018-08-03:
You can find the right value in the header of your log:

Example in a log coming from a FC using a MPU6000:
`H acc_1G:2048`

col85, 2018-08-03:
@shellixyz  I was also tracing the codes and realised that sysConfig.acc_1G is needed in the computation for actual acc in g or m/s^2. Thanks so much for also looking into this. I guess in this case gyroADC values conversion should be of similar approach.

shellixyz, 2018-08-03:
For the gyro the values are directly in °/s

col85, 2018-08-03:
Was going through the functions in blackbox_decode.c, I think the default values for gyro are the raw values too. There is a function inside (flightlogGyroToRadiansPerSecond(log, fieldValue) * (180 / M_PI) to convert them to degrees/s. Is my understanding correct or have I mistaken them again?

shellixyz, 2018-08-04:
Looks like they are the raw values in the log but the blackbox_decode tool converts the value to °/s too. Like the viewer does.

col85, 2018-08-04:
Sure, thanks so much for your help in this! Going to close this since its cleared up. Cheers!

closed 2018-08-04 by col85