iNavFlight/inav #3692: BUG: 2.0.0RC4 Incorrect OSD RSSI value if transmitter is power off and use RSSI channel (like channel 16)

davidngrc, 2018-08-03:
Actually, I am no the latest development build, just compile it.

For R9 mini, and others, it does not come with a analog RSSI pin/pad.
so it use transmitter (X9D) channel 16 as RSSI value.

if power off your transmitter, you will see in OSD, the RSSI value is 49
which is not correct, it should be 0?

shellixyz, 2018-08-03:
It is displaying 49 when you power up the board without the transmitter on is because the default value of the channels is 1500 and is is still displaying 99 when you power it off because the channel is frozen when in failsafe.

Maybe we should display 0 when in failsafe ?

davidngrc, 2018-08-04:
yes
1) in FS, set the raw RSSI channel value to be 0
or
2) when output to the OSD, then use if RSSI channel is set, then set the OSD rssi value to 0
which one is better?

digitalentity, 2018-08-04:
Caused by the fact that in 2.0 `rcData` is no longer updated if receiver is in failsafe. Fix pending.

digitalentity, 2018-08-04:
Fix ready. Closing

closed 2018-08-04 by digitalentity