iNavFlight/inav #3697: External BMP280 barometer not working.

ninja-zx11, 2018-08-05:
### Board and Version
OmniNXTF7
INAV/OMNIBUSF7NXT 2.0.0 Jul 14 2018 / 22:41:52 (f2cf315ad)

### Behavior
External Bmp280 connected to i2C bus doesn't work.

The inbuilt barometer ST LPS25HB doesn't look precise.It comes up as 'FAKE' in the configuration menu.It works but not as precise as BMP280 or MS5611.So i tried to install external BMP280 module on SCL and SDA but it's not working and showing up as Red.No sensor readings.

Diff:
`# diff

# version
# INAV/OMNIBUSF7NXT 2.0.0 Jul 14 2018 / 22:41:52 (f2cf315ad)

# resources

# mixer

# servo mix

# servo

# feature
feature GPS

# beeper

# map

# serial
serial 5 2 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# osd_layout

# master
set acc_hardware = MPU6000
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set gps_sbas_mode = WAAS

# profile
profile 1

# battery_profile
battery_profile 1

# 

`
[dump.txt](https://github.com/iNavFlight/inav/files/2260041/dump.txt)

Tygrys-1, 2018-08-05:
Is I2C properly terminated/does it have pullup resistors?

ninja-zx11, 2018-08-05:
I m not sure but I tried connecting external bmp280 baro, OLED screen and external hmc5883L compass all at same time and both OLED and compass are working.My bmo280 module is tested working on F3 board.

Tygrys-1, 2018-08-06:
Maybe check for I2C adress conflict? Devices sometimes try to mimic other devices and you cannot have two devices with the same I2C adress.

ninja-zx11, 2018-08-06:
How can i check I2C address conflict?Sorry i am not an expert but can do with guidance.Thanks

Tygrys-1, 2018-08-06:
Each I2C devide has it's own unique adress. For BMP 280 it is AFAIR 7A or 79. If the emulated Baro has the same adress the external one it will be a conflict.

ninja-zx11, 2018-08-06:
After searching i found that LPS25HB I2C address is 0x5C(92).So it should not conflict.Unless the baro is fake and it has a I2C address same as BMP280. 

Another question:Why Inav configurator is showing baro as 'FAKE' if its LPS25HB?but if i use status command then it says LPS.

stronnag, 2018-08-06:
Because the configurator has not caught up with the firmware. Raising an issue on the configurator github may accelerate update.

ninja-zx11, 2018-08-06:
Ahh! Ok I will create an issue on Inav configurator.
Is there any way to look up inside a firmware for the programed address of Lps baro?I don't know how it works but could it be that omninxtF7 has 2 i2c busses and it's not programed to see external barometer on second i2c bus as it has inbuilt baro?

crashfrequently, 2018-08-06:
I'm no expert but (in target.h) it seems that only the I2C1 bus is defined and that only supports mag, pitot and rangefinder.
No support for any other baros other than the built in LPS25H on the SPI bus.