iNavFlight/inav #3699: New target for Omnibus F4 clones possible?

Ambrotos, 2018-08-05:
I have a few "Omnibus F4 2.1" that I ordered from Banggood. I've been successfully using Betaflight 3.3 and 3.4 on it, but I'm messing around with building a new PTZ camera platform and I'd like to use iNav now. Unfortunately, for the life of me I can't get any of the recent 2.0.0 RCs to work properly on it. The problem is that my Spektrum Serial RX receiver will not work at all. I've done a fair bit of Googling and searching through this issues forum, and it seems like there are many others also having trouble with serial Rx on these clone boards, but the advice given didn't seem to apply to my situation.

In my case, I'm using a Spektrum Serial Rx plugged into the 3-pin DSM port (adjacent to the USB port), which I know is just another hookup to UART1Rx with a 3.3v power lead. And I know that Betaflight 3.4 works properly in this configuration, specifically using the "OMNIBUSF4SD" target. Checking the iNav wiki, the recommended target for these BG clone boards seems to be iNav's OMNIBUSF4PRO, though I've also tried the V3 target with no difference. I figured maybe it was a problem in the way iNav was initializing the UARTs. So I opened up the ../OMNIBUSF4/target.h file and modified the way it was initializing UART1 for the OMNIBUSF4PRO build target to more closely match the way it's done in Betaflight 3.4. 

To be specific, I:
- Commented out the #define UART1_AHB1_PERIPHERALS  RCC_AHB1Periph_DMA2 on line 128 (honestly not really sure what that does)
- Commented out the #define INVERTER_PIN_UART1 on line 47.

Then build and load the image onto my FC. Lo and behold, my Serial Rx works now on the DSP port!!

So, I guess this is a long winded way of asking whether there is a change needed to the OMNIBUSF4PRO build target for UART1 initialization, or if this is just a difference between clones and non-clones. If it's a clone-only issue, is there a chance we could get a new build target? I'd be happy to provide my changes, though I was really just hacking around to see if I could identify the issue. I could also test any proposed changes made by people who actually know what they're doing, since I'm assuming not everyone has one of these FCs kicking around ;)

Thanks,

-A

shellixyz, 2018-08-06:
Did you try changing the `serialrx_inverted` setting ?

Ambrotos, 2018-08-07:
I hadn't actually tried that setting, no. I think I remember reading somewhere (possibly in an Oscar Liang post?) that one of the differences between F3 and F4 controllers is that the F4 STM32 chip doesn't have the ability to do runtime UART inversion, which is why I started going down the custom build route. As it turns out I must have misunderstood what I read because that setting seems to have done the trick! Using the latest 2.0RC5 release I was able to get my Rx working on the DSM port by simply setting serialrx_inverted = ON.

Thanks a ton for the tip! I usually try to do my homework before posting on support forums, but in this case I wish I'd posted sooner. A much simpler solution than I'd expected ;)

Cheers,

-A

teckel12, 2018-08-07:
@Ambrotos You read correctly, the F4 can't invert a UART internally like with the F3 and F7.

But...  On some F4 boards you can turn on and off the inverter circuit on the FC board via a pin.  Some boards have a solder pad to turn inversion on/off and others have both inverted and non-inverted pins exposed.

It sounds like you have a board with the inverter circuit wired to a pin that you can control to turn on/off.

smaller09, 2018-08-07:
take a look at  #3610 , U have to unsolder a resistor label R20 to get dsm sat to work, or U can wait RC5(maybe) to correct it #3673  or just pitch the sources

sq9fk, 2018-09-07:
Take a loot at my response: 
https://github.com/iNavFlight/inav/issues/3574#issuecomment-419371066