iNavFlight/inav #3706: Strange behavior in POSHOLD Cruise mode when doing yaw turns

DiscoMan18, 2018-08-06:
Matek F405-CTR, iNav 1.9.1, GPS Radiolink TS100 mini 
650-sized quadcopter, 6S battery, 15 inch props, 320kv motors, AUW 1.9 kg.

In ANGLE and ALTHOLD modes my quad flies just great, it's totally controllable and predictable. In POSHOLD mode it hold the position very well, correctly responds to stick inputs, rotates by yaw axis while staying exactly at one point without any drift etc. There are usually enough satellites in use, 12 or more, and HDOP is generally less than 1.8.

When flying in POSHOLD Cruise mode at relatively low speeds (15-20 kmh) without doing sharp yaw turns, everything looks good too, the quadcopter slightly tilts itself inside the turn and returns to level position at the end of the turn. (is it a normal behavior btw?) Strange things start happen when I try to do sharper yaw turns at higher speeds. Now the quad tilts itself much much more, not only on roll axis, but also in pitch. At this moment angles may become really weird: 20-30 degrees or more. Then it tries to "compensate" and now tilts to opposite direction making me fear that it could become totally uncontrollable. Here I usually switch off POSHOLD in panic and the quad immediately calms itself down. 

What could cause this behavior? A bug? A compass issue? Or navigation PIDs tuning issue? 

Also, does anybody know what variables in blackbox log correspond to current heading reported by magnetometer and by GPS? 

Blackbox logs and settings are below. In logs the behavior described above start from approx 04:00 and ends at 04:34 when I switched POSHOLD off and released the sticks. 

`
# version
# INAV/MATEKF405 1.9.1 Apr 21 2018 / 13:25:29 (03a5c1922)

# reset configuration to default settings
defaults noreboot

# resources

# mixer

# servo mix

# servo

# feature
feature GPS
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper -RX_LOST
beeper -RX_LOST_LANDING

# map
map TAER

# name

# serial
serial 0 0 115200 38400 0 115200
serial 3 2 115200 57600 0 115200
serial 4 16 115200 38400 1200 115200

# led

# color

# mode_color

# aux
aux 0 1 0 900 2100
aux 1 3 0 1425 1700
aux 2 9 0 1775 2000
aux 3 18 0 900 1150

# adjrange

# rxrange

# master
set gyro_sync = ON
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 55
set gyro_notch1_hz = 105
set gyro_notch1_cutoff = 90
set gyro_notch2_hz = 72
set gyro_notch2_cutoff = 50
set acc_hardware = MPU6000
set align_mag = CW270FLIP
set mag_hardware = QMC5883
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 8
set min_throttle = 1100
set max_throttle = 1950
set failsafe_delay = 10
set failsafe_off_delay = 0
set failsafe_throttle = 1200
set failsafe_throttle_low_delay = 0
set failsafe_procedure = RTH
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 260
set vbat_warning_cell_voltage = 270
set current_meter_scale = 185
set small_angle = 180
set disarm_kill_switch = OFF
set gps_sbas_mode = AUTO
set gps_min_sats = 5
set deadband = 7
set alt_hold_deadband = 20
set nav_use_midthr_for_althold = ON
set nav_user_control_mode = CRUISE
set nav_wp_radius = 200
set nav_auto_speed = 1250
set nav_auto_climb_rate = 300
set nav_manual_speed = 1500
set nav_manual_climb_rate = 400
set nav_emerg_landing_speed = 150
set nav_rth_altitude = 4000
set nav_mc_bank_angle = 25
set nav_mc_hover_thr = 1480
set osd_time_alarm = 120
set osd_alt_alarm = 1000
set osd_main_voltage_pos = 2081
set osd_rssi_pos = 2233
set osd_flytimer_pos = 2444
set osd_flymode_pos = 2093
set osd_throttle_pos = 2337
set osd_vtx_channel_pos = 408
set osd_artificial_horizon_pos = 200
set osd_horizon_sidebars_pos = 200
set osd_current_draw_pos = 2087
set osd_mah_drawn_pos = 2145
set osd_gps_speed_pos = 2098
set osd_gps_sats_pos = 2169
set osd_gps_lon_pos = 2481
set osd_gps_lat_pos = 2468
set osd_home_dir_pos = 2159
set osd_home_dist_pos = 2105
set osd_altitude_pos = 2137
set osd_vario_num_pos = 2297
set osd_power_pos = 2113
set osd_ontime_flytime_pos = 375
set osd_messages_pos = 2497
set osd_gps_hdop_pos = 2201
set osd_throttle_auto_thr_pos = 2369
set osd_efficiency_mah_pos = 2177
set async_mode = GYRO

# profile
profile 1

set mc_p_pitch = 45
set mc_i_pitch = 15
set mc_d_pitch = 50
set mc_p_roll = 45
set mc_i_roll = 15
set mc_d_roll = 50
set mc_p_yaw = 70
set mc_i_level = 7
set dterm_lpf_hz = 30
set dterm_notch_hz = 108
set dterm_notch_cutoff = 72
set rate_accel_limit_roll_pitch = 10000
set rate_accel_limit_yaw = 250
set nav_mc_pos_z_p = 55
set nav_mc_pos_z_i = 18
set nav_mc_vel_z_i = 95
set nav_mc_vel_z_d = 15
set nav_mc_pos_xy_p = 15
set nav_mc_pos_xy_i = 100
set nav_mc_vel_xy_p = 25
set nav_mc_vel_xy_i = 20
set nav_mc_vel_xy_d = 50
set thr_mid = 40
set thr_expo = 30
set rc_expo = 60
set rc_yaw_expo = 35
set roll_rate = 30
set pitch_rate = 30
set yaw_rate = 14

# profile
profile 2

# profile
profile 3

# restore original profile selection
profile 1

# save configuration
save

# `

Blackbox logs: [LOG00053.TXT.zip](https://github.com/iNavFlight/inav/files/2263947/LOG00053.TXT.zip)

giacomo892, 2018-08-07:
Try to fly enabling TURN_ASSISTANT mode together with POSHOLD. It should help in sharp turns.
Also set rate_accel_limit_yaw = 250 is quite low to be honest. The max bank angle is set to 25, so it cannot go over that angle.

DiscoMan18, 2018-08-08:
The problem is that I have one more quadcopter, exactly of the same config and with exactly the same settings, that flies great. And at the same I have one more quad)) this time of 450 size and flashed with inav 2.0 rc4 which behaves erratically, just like the first quad. Could somebody take a look at the logs? From my point of view logs look ok, I couldn't find any signs of compass or vibration issues...

DiscoMan18, 2018-08-29:
During past days I tried 1.9.1 and 2.0 release firmwares, tried two different GPS modules (ts100 mini and Naza 2 GPS) - this "strange behaviour" is still present. The quadcopter hovers steadily at one point, rotates by yaw input without any drift, RTH works perfectly and in general it flies great in Poshold attitude mode.

Here is the video demonstrating the issue:

[https://youtu.be/c_ge14j56AQ](url)

DiscoMan18, 2019-01-08:
> This is likely a tuning issue. Either tune for sharper response in ANGLE mode or tune to loosen response on VelXY. Could also be aircraft physical limitation.
> 
> When VEL PID doesn't see the quad change speed as desired - it will push more (tilt more). If response is slow (which is expected on machine of this size) quad is likely to overshoot eventually.
> 
> I'd start with PosXY/VelXY PIDs first.

Actually I spent the whole summer trying to figure out what's wrong with my quad, is it a tuning or a firmware issue. I also tried using default PosXY/VelXY pids, which are way more "agressive" than mine. No luck. 

Also, "machine size" is not that big. As I said above, I got the same issues on all my quads, equipped with iNav, from 450-sized quad with 9" props to 700-sized quad with 15" props.

closed 2019-03-23 by stale[bot]