iNavFlight/inav #3707: Can I use the Alt Hold mode for gps? (not barometer)

whakru, 2018-08-06:
Can I use the Alt Hold mode for gps?
Without barometer, only gps
Matek f722 - I spoiled my barometer when I protected the board.
I would like to have Alt Hold in height, which gives gps

shellixyz, 2018-08-07:
Only for planes, not multirotors.

whakru, 2018-08-07:
What about these values?
[image]

digitalentity, 2018-08-07:
Multirotor altitude hold requires barometer.

closed 2018-08-07 by digitalentity

golffer, 2018-12-05:
what's recommend number for these 2 weight? QAV 250, Omnibus F4 V3.
When it is 0.35 and 0.2, quadcopter jump up. 
When it is 0.5 and 0, quadcopter drop slowly to ground.
Gonna crazy....

giacomo892, 2018-12-05:
Leave to default and isolate the barometer better @golffer 
Changing numbers randomly won't help