iNavFlight/inav #3716: HDOP reported by iNav is much higher than that read by uCenter

johnsobr, 2018-08-08:
Matek Wing, purchased from GetFPV 6/21/2018

INAV/MATEKF405SE 2.0.0 Jul 14 2018 / 22:38:02 (f2cf315ad)
RC4

iNav OSD reports much higher HDOP than uCenter
Expected iNav OSD HDOP to match what uCenter says

HDOP reported:
uCenter   iNav

0.8            1.7  (see screen shot photo)
1.7            2.3    
0.7            1.1
0.7            1.2
0.8            1.3
0.8            1.4

[image]

I reset factory defaults in the GPS and I enabled Glonass in iNav (why this is OFF by default, I don't know) using the CLI get gps_ublox_use_galileo to verify it was in fact OFF, then set it to ON

Drotek GPS with a USB port for configuring, USB works simultaneously while iNav is communicating at 57,600 to the GPS via serial port.

U-Center on my PC reports HDOP 0.8 while at the same time iNav is showing HDOP 1.7 on the OSD. Thats after a lot of settling time, and its outside looking at a clear sky, with the GPS out on a wing tip on long extension wires to get it away from electronics, and USB to my HP desktop ;<)

This makes it very frustrating waiting for the NAV NOT POSSIBLE message to clear since it thinks I have too low of an HDOP for much longer than necessary.

[HDOP Dump.txt](https://github.com/iNavFlight/inav/files/2268492/HDOP.Dump.txt)

[image]
[image]

more details here:  https://www.rcgroups.com/forums/showthread.php?1518533-NEW-DRAGON-LINK-ADVANCED-Full-Telemetry-Bluetooth-Droid-Planner-Data-Modem/page4855

teckel12, 2018-08-08:
@johnsobr Others have reported that maybe INAV shows PDOP instead of HDOP values. But, these numbers are  arbitrary anyway, so I'm not sure it makes much difference. Numbers less than about 3.5 are good enough for RTH, less than 2.0 is good for stable position hold.

If you research GPS DOP, you'll see all kinds of different techniques. Some better than others. But basically the lower the better, but don't hold too much value in the actual number.

stronnag, 2018-08-08:
See also https://www.rcgroups.com/forums/showpost.php?p=39837922&postcount=20378

teckel12, 2018-08-08:
@johnsobr UBLOX GPS modules report PDOP, VDOP, TDOP, and HDOP. From the UBLOX documentation: 

> PDOP - numeric - Position dilution of precision
> HDOP - numeric - Horizontal dilution of precision
> VDOP - numeric - Vertical dilution of precision
> TDOP - numeric - Time dilution of precision

Looking at the UBLOX code in INAV, it appears INAV is getting the PDOP and using that for the HDOP value.  See the following in `gps_ublox.c`:

```C
gpsSol.hdop = gpsConstrainHDOP(_buffer.solution.position_DOP);
```

This is not such a bad thing.  As HDOP is just the horizontal dilution of precision.  You can think of PDOP as a 3D DOP. It's also referred to as GDOP (geometric dilution of precision).

In any case, INAV appears to be using the PDOP value from the GPS, which is a good thing as it's a better measurement of location accuracy.  It's just referring to it as HDOP.  This is probably based on old code from old GPS modules that only reported HDOP.

If you look at your screen shots.  The PDOP value was 1.6 while the OSD was showing 1.7.  This confirms what INAV is doing.  So, while INAV labels it as HDOP for legacy reasons (and probably because people are more familiar with the HDOP label), it's really PDOP, which is what you should be looking at in uCenter anyway as that's a better measurement of location accuracy than HDOP, which is only looking at horizontal dilution of precision.

Maybe we should change the OSD's label to be PDOP instead of HDOP if using UBLOX or NAZA?

digitalentity, 2018-08-08:
Or maybe just "DOP" to save some space on screen 😄

teckel12, 2018-08-08:
Reviewing the GPS code and spec sheets.  UBLOX and NAZA are using PDOP and i2cnav and nmea are using HDOP.  Isn't i2cnav removed in 2.0 RC5?  Or was that just for F1 boards?  If so, only nmea is using HDOP.

I guess changing the label on the OSD and configurator to be DOP would be technically correct, but it may not correct any confusion, or may create more.

johnlenfr, 2018-08-28:
And why not calculating a "3D DOP" or a "RMS" by using (VDOP^2+HDOP^2)^1/2 ?

teckel12, 2018-08-28:
@johnlenfr PDOP **is** 3D DOP.  Which is why for UBLOX and NAZA we use the PDOP value (PDOP is the other name for 3D DOP). If anything, we should be usig PDOP and TDOP and generating GDOP to add time dilution of precision in the mix.

But honestly, PDOP (3D DOP) that  UBLOX and NAZA are outputting as HDOP in INAV is good enough as the numbers really doesn't relate to absolute accuracy anyway.

johnlenfr, 2018-08-28:
@teckel12 Ok thanks

johnsobr, 2018-08-28:
HDOP – horizontal dilution of precision
VDOP – vertical dilution of precision
PDOP – position (3D) dilution of precision
TDOP – time dilution of precision
GDOP – geometric dilution of precision

digitalentity, 2018-08-28:
xDOP values are really arbitrary as they reflect accuracy relative to some module-specific reference. Thus it doesn't matter which one we show. PDOP (3D-DOP) seems most meaningful.

closed 2018-08-28 by digitalentity