iNavFlight/inav #3719: 2.0RC5, target KAKUTEF4V2 sbus not working

DaPeace, 2018-08-08:
Ive upgraded from 1.91 to 2.0RC5 today but the sbus-receiver is not really working. Ive completely wiped the flash and tried to configure the sbus-receiver directly after flashing and its the same behavior.
There is a big delay on inav navigator when you move the sticks or enable/disable buttons, when connecting via mobile flight the rx-feed is not detected and its not possible to arm the drone.
When reflashing the drone with 1.91 everything is fine again. Ive tried different RCs and its all the same.

digitalentity, 2018-08-08:
Which receiver do you use?

DaPeace, 2018-08-08:
https://www.aliexpress.com/item/MOY-RXA2-10-2-4G-10CH-AFHDS-2A-Mini-Receiver-with-PPM-SBUS-Output-for-Flysky/32832578891.html?spm=a2g0s.9042311.0.0.5b2f4c4d5xMQSr

At the moment ive connected this one to RX3.

digitalentity, 2018-08-08:
@DaPeace try reducing `sbus_sync_interval` from default 3000 to 1000 or 500 and see if it makes the difference. If receiver is not respecting SBUS specs 2.0 firmware will drop SBUS frames it considers invalid.

DaPeace, 2018-08-08:
Ok. did that change from 1.91 to 2.0? I will try out tomorrow.

digitalentity, 2018-08-08:
2.0 introduced a lot of checks for S.Bus protocol to avoid corrupt frames to leak through and cause issues. Apparently some receivers (Radiolink for sure) don't respect S.Bus timings.

DaPeace, 2018-08-08:
Ok. Thanks for the information. The moy RXA was only interessting because of the 10channels. Ill try to lower the settings and if i dont get it running i will try my turnigy-receivers. Maybe those are better.

DaPeace, 2018-08-09:
set sbus_sync_interval = 1500 did the trick. Thanks for help!

DzikuVx, 2018-08-09:
closing then. Thanks

closed 2018-08-09 by DzikuVx

Fredisk, 2019-06-01:
having the same problem with R9SLim+, with every FC