iNavFlight/inav #3727: Omnibus F4 Pro 2.1 UART6 SBUS - new build target needed?

modellbobby, 2018-08-10:
I too had the issue with my bang good Omnibus F4 Pro 2.1 and when flashing the target **omnibusF4pro** **ledstrip** the receiver UART6 did not work. Only the OmnibusF4V3 target worked now but 1) I am not sure If this is correct 2) the led strip on PWM 5 is not working.

Do I need to solder the two tiny pads left from PWM1 to SBUS to get the inverter running?
Just wondering why the target Omnibus F4 V3 works here..

Help appreciated :)

teckel12, 2018-08-10:
The pro model is the one with the current sensor. I've seen clones called a pro when it isn't. Even then, there's so many omnibus clones out there it's almost impossible to make everything work for everyone.

Maybe you could show a link to your board? If your using SBUS, need the signal inverted, and there an SBUS pad selection, you need to solder that connection to enable the inverter.

Anyway, a picture or link would help make sure what hardware you have.

modellbobby, 2018-08-10:
Hi,

this is the strange board:

https://www.banggood.com/BF3_1_5-Omnibus-F4-V2-Flight-Controller-STM32-F405-MCU-Integrated-OSD-Built-in-5V-BEC-Current-Meter-p-1143258.html?rmmds=myorder

teckel12, 2018-08-10:
@modellbobby If that's the board you have, it's a v2 clone with the classic club24a problem.  It's probably named v2.1 so you don't think it has the club24a problem, but it probably does.  This board has had classic problems and should be avoided.  I would really suggest getting an ACTUAL Omnibus F4 Pro from racedayquads.com or readytoflyquads.com.  Those are the only two places I know of that sell non-clone boards, but there could be others.  Banggood and other Chinese stores list Omnibus F4 Pro V3 boards, but I just found a few that are still the old V2 with the club24a problem like yours.

Even if you can get your board to work, I can almost guarantee it's prone to the club24a problem.  Good luck!

modellbobby, 2018-08-11:
hello, the only issue I have is that the OMNIBUSF4PRO target is not working because the UART6 needs an inverter for SBUS. The OMNIBUSF4V3 target works now but the LEDSTRIP does not work in PWM5.

teckel12, 2018-08-11:
@modellbobby I'd look into the club24a problem if I were you. I personally would not fly this FC at all. 

Did you set the SBUS inversion with the pads? Also, clones can be wired totally differently. I'd research the target differences between BF and INAV. INAV has the OMNIBUSF4PRO target setup for the v3 model (as no one should be using the highly problematic v2 like yours). The other targets may be close, but could have some differences. You may need to create a custom target for a clone board.

closed 2018-10-24 by stale[bot]