iNavFlight/inav #3728: BMP280 baro does not work on BETAFLIGHTF4

trekki1978, 2018-08-11:
Hi,
I wanted to install inav on my BETAFLIGHTF4 board. All works fine but the baro BMP280 connected to TX3/RX3 (=I2C) is not working. On betaflight it works. So I had a look at the following file:
https://github.com/iNavFlight/inav/blob/master/src/main/target/BETAFLIGHTF4/target.h

In lines 54 and 55 the following definitions are set:
#define BMP280_SPI_BUS		BUS_SPI2
#define BMP280_CS_PIN PB3

In the lines 135 and 136 the following is set:
#define USE_BARO
#define BARO_I2C_BUS BUS_I2C2

In the same file on github/betaflight I could not find these definitions of lines 54 and 55. Why are these necessary on iNav? Is it possible that BMP280 does not work because of "#define BMP280_SPI_BUS" and "#define BMP280_CS_PIN PB3"?

Many thanks
Marco

digitalentity, 2018-08-12:
Interesting. IIRC BetaflightF4 board used to have an onboard baro on SPI bus.

teckel12, 2018-08-12:
@digitalentity I know I've seen pictures of the Betaflight F4 board with a barometer on-board. But now I'm reading that there isn't one. It would be an awesome board with baro.

trekki1978, 2018-08-12:
Hi,
was a little bit confused to read here that there would be a BFF4 board with onboard baro. Now I have found two different pictures showing the same board without baro:
[image]
and with a baro:
[image]

Unfortunately I have the one without barometer.
Maybe it's not a good idea to remove the SPI bus definition if there are two differrent boards. Is it possible to release two different firmwares?

If I had the necessary experience, I would just try to compile a version for me without SPI bus definition to see if it works...

teckel12, 2018-08-13:
@trekki1978 The low silkscreen quality of the one without barometer makes me wonder if we're looking at a clone.

trekki1978, 2018-08-13:
I was also wondering if the one or the other is a clone. But as the Betaflight firmaware does not include a definition for an internal baro on the SPI bus, I would think that the board with baro is a clone?!

trekki1978, 2019-01-27:
Hi,
I was waiting for the fixing of this issue a while. Now I have fixed this issue by myself by adding "//" to the lines 51 and 52 in the target.h file and compiling a new .hex file. And this seems to work.
Am I allowed to change the target.h file in github or does it have to be done by a contributor.

digitalentity, 2019-01-29:
@trekki1978 making that change will break support for built-in baro on board that have it. The correct way of fixing it is making it possible to chose between external/internal baro in a way similar to gyro selection on dual-gyro box. Not sure if we'll be able to do it for 2.1 though.

ulph0, 2021-04-24:
hi, before to post an unqualified issue i had a discussion with the brave 'snowleopard' here: https://intofpv.com/t-newbies-question-belonging-to-flight-controller and finally i found this post! what a luck! you have a bit confusion with line numbers above. also INAV is on v2.6.1 now. i also commented out the two lines 
#define BMP280_SPI_BUS BUS_SPI2
#define BMP280_CS_PIN PB3
by adding '//' to the begining of the line, compiled your code for the first time and it worked like a charme. of course it would be much nicer if it would be possible to configure this easier than through compiling the code of each new version on myself until the end of time.
my FC is an iflight Succex-E F405 V1. it has also no intern baro, while version 2 of the FC has. problem is that i ordered V2 on banggood but what arrived was a V1, it seems they deliver erratically one of that two versions when you order the stack with ESC.

closed 2022-05-07 by DzikuVx