iNavFlight/inav #3730: F405STD no GPS Fix

HJOGermany, 2018-08-12:
Hello

i have a new F405STD Board connectes with a BN880 GPS.
In the Setup Tab the Icon ist blue an in the GPS Tap i see that there is Traffic between GPS an FC.

But i cant get a GPS Fix. Waiting 15min (Cold Start) on my Field but nothing.
Inav 1.9.3 and newer Versions.

GPS tested in my Wing, its OK

Can somebody help me?

`
# version
# INAV/MATEKF405 1.9.1 Apr 21 2018 / 13:25:29 (03a5c1922)

# resources

# mixer

# servo mix

# servo

# feature
feature -TX_PROF_SEL
feature SOFTSERIAL
feature GPS
feature RSSI_ADC
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# name
name Chamäleon

# serial
serial 1 0 115200 38400 0 115200
serial 2 64 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 4 1024 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1300 1700
aux 1 1 3 900 1300
aux 2 2 0 1300 1700
aux 3 9 8 900 1300
aux 4 8 8 1700 2100
aux 5 19 5 900 1300
aux 6 11 2 925 1325
aux 7 29 6 900 1300
aux 8 30 0 900 1300
aux 9 31 7 900 1300

# adjrange

# rxrange

# master
set looptime = 2000
set gyro_sync = ON
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 90
set gyro_notch1_hz = 200
set gyro_notch1_cutoff = 100
set acc_hardware = MPU6500
set acczero_x = 1
set acczero_y = -57
set acczero_z = 17
set accgain_x = 4089
set accgain_y = 4082
set accgain_z = 4090
set mag_hardware = HMC5883
set magzero_x = -42
set magzero_y = -108
set magzero_z = -313
set baro_hardware = BMP280
set pitot_hardware = NONE
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set motor_pwm_rate = 2000
set motor_pwm_protocol = MULTISHOT
set failsafe_procedure = RTH
set vbat_max_cell_voltage = 420
set gps_sbas_mode = AUTO
set gps_ublox_use_galileo = ON
set inav_w_z_baro_p =  0.350
set nav_extra_arming_safety = OFF
set nav_rth_tail_first = ON
set nav_rth_allow_landing = FS_ONLY
set nav_rth_altitude = 3000
set osd_video_system = 2
set osd_main_voltage_pos = 2091
set osd_rssi_pos = 2105
set osd_flymode_pos = 2413
set osd_throttle_pos = 65
set osd_horizon_sidebars_pos = 200
set osd_current_draw_pos = 97
set osd_mah_drawn_pos = 129
set osd_gps_sats_pos = 2082
set osd_gps_lon_pos = 2433
set osd_gps_lat_pos = 2401
set osd_home_dir_pos = 2254
set osd_altitude_pos = 2209
set osd_ontime_flytime_pos = 2421

# profile
profile 1

set mc_p_pitch = 58
set mc_i_pitch = 50
set mc_d_pitch = 22
set mc_p_roll = 43
set mc_i_roll = 40
set mc_d_roll = 20
set mc_p_yaw = 70
set dterm_lpf_hz = 80
set dterm_notch_hz = 200
set dterm_notch_cutoff = 100
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 45`

digitalentity, 2018-08-12:
Are you using Runcam Split camera? How is your GPS mounted? Please send a picture of your setup

HJOGermany, 2018-08-12:
Yes, I use a V1.
The GPS ist mounted at the place where normally mounted a external Cam.
[image]

digitalentity, 2018-08-12:
Most action cameras are known to mess up GPS
https://github.com/iNavFlight/inav/issues/3643#issuecomment-408101542

HJOGermany, 2018-08-12:
Ok my Englisch is Not so Good that I understand the posts in your link.
It is right that you mean the Split makes that the GPS isn't running?
Can I save it with Alloy foil or must I solder out the Split?

shellixyz, 2018-08-13:
Your GPS in the picture is upside down. The antenna needs to point to the sky. The grey/pale pink/silver square. The first thing is to fix this.

For shielding you can try with some mumetal foil if you can find some, not sure it would help. The best would be to isolate the Split electrically (separate BEC) and also put the GPS as far away as possible from it.

teckel12, 2018-08-13:
@shellixyz Good catch.

HJOGermany, 2018-08-13:
@shellixyz: Thank You 
That was it 😁

I cut off the power and after 2mins I have 3D Fix

Now I must test how I can shield the Split.

Thank You

closed 2018-08-13 by HJOGermany