iNavFlight/inav #3731: OMNI NXT F7 orientation problem

CcLaFleche, 2018-08-12:
Hi,
Thanks for your work but there is a problem with the orientation of this board.

CcLaFleche, 2018-08-13:
I have to put cw180 accelero and mag for have correct mouvements in first menu in the configurator but when i armed the multirotor turn down and broke my motors and props.

When i calibrate the accelerometer the figure valided is not the same position of my fc, even if i do it with the default values.

digitalentity, 2018-08-13:
@CcLaFleche generally you shouldn't touch `align_acc` and use board alignment. By blindly adjusting `align_acc` you created a misalignment between gyro and accelerometer which is the root cause of your issue.

CcLaFleche, 2018-08-13:
it's the same with the original settings, hence my approach.
I think the original alignment is not good ...
can we choose one of two gyro because I did not succeed.
can we also use an external baro

digitalentity, 2018-08-13:
The board doesn't have the direction arrow. Make sure you don't touch acc/gyro alignment blindly. Boards that have onboard baro generally can't use external one. You can chose one of two gyros using `gyro_to_use` CLI setting on boards that support the feature.

CcLaFleche, 2018-08-13:
how can I do so that the controller turns in the same direction as in the configurator WITH the original settings because this is not the case.
With betaflight I do not have this problem.

If I doused the baro, I can use another on scl and sda

the second gyro (32k compatible) doesn't work

CcLaFleche, 2018-08-13:
If I doused the baro, I can use another on scl and sda ?

digitalentity, 2018-08-13:
Baro on NXT board is on SPI bus. You can't connect another one to I2C

CcLaFleche, 2018-08-13:
Pour resumer :
il me faut utiliser un controleur de vol dépourvu de baro mais possédant un I2C

CcLaFleche, 2018-08-13:
I have to use a flight controller without a baro but with an I2C

CcLaFleche, 2018-08-13:
Like images are better than words ...
I made 2 videos to illustrate my subject which does not seem to be understood.

https://www.youtube.com/watch?v=Fyg5QAb1zJ8

https://www.youtube.com/watch?v=X7NTtCQ53zQ

digitalentity, 2018-08-13:
As indicated before - read the docs and set `align_board_yaw` properly. Prior to doing that make sure acc and gyro alignment are at defaults.

CcLaFleche, 2018-08-13:
i will test 180 yaw for board and fly.
I will comfirm it to you
Could you add programmation for use and external baro for better sensor as BMP280 ?
Why do not use dshot on inav ?

CcLaFleche, 2018-08-31:
It's ok...

closed 2018-08-31 by CcLaFleche