iNavFlight/inav #3741: Remove acc/gyro alignment settings

digitalentity, 2018-08-13:
Those components are fixed on board and their orientation really can't be changed. The only case where different alignment is possible - is when using different sensors on a dual-gyro board.

GiorgosVasia, 2018-08-17:
I have a matek f-405 std and i fly with inav 1.9 since it released.all features work perfectly.I couldnt align my board with align_board and the only thing that worked for me was to use align_gyro and align_acc commands.So i think the commands shouldnt be removed.

digitalentity, 2018-08-17:
@GiorgosVasia what's the problem with using `align_board_yaw`?

b14ckyy, 2019-06-11:
> @GiorgosVasia what's the problem with using align_board_yaw?

Maybe he talks about this: 

> Attention
> After flashing BF4.0, MUST set F405-STD & F405-mini first gyro to “CW180” in configurator to follow the arrow forward 

Source http://www.mateksys.com/?portfolio=f405-std

not sure if this counts for INAV too.

closed 2022-05-07 by DzikuVx

zgliyun, 2022-05-23:
I think this feature should be kept.
My iflight F745 AIO have different acc & gyro alignment, which in default, gyro alignment is correct but acc should use CW90.
It use BMI270 and I not sure if other BMI270 board have  the same issue.
Finally I set align_gyro = DEFAULT & align_acc = CW90 & align_board = DEFAULT. Then everyting went perfect. 
I try to modify target.h to resolve this problem, but can't.

DzikuVx, 2022-05-23:
@zgliyun which INAV version? Because in 4.0 was a bug that might have caused such a behavior on BMI270. It'd fixed in 5.0

zgliyun, 2022-06-03:
4.1.0
I use set align_acc = CW90 to correct this.

zgliyun, 2022-06-03:
Let me try 5.0.0 first.