iNavFlight/inav #3742: omnibus f4 v5 custom mix

nanoperry, 2018-08-14:
i have a nano talon with a omnibus f4 v5
i use inav but i cant get it to work with custom mix
if i click on a wing or aircraft everything is working
i use inav 2.0 rc4 but the firmware update is from 22-4-2018 1.9.1
and when i click in inav on the mix tab then he say that my firmware must be updated
and there is no newer firmware in the list
can i please get some help

sorry for my english

DzikuVx, 2018-08-14:
You have to update both Configurator and flight controller firmware to 2.0 to use Mixer tab https://www.youtube.com/watch?v=sHvItS2qKfk

closed 2018-08-14 by DzikuVx

nanoperry, 2018-08-14:
yes but when i look at the new firmware the newest is 1.9.1 and when i do this and click on mix tab inav say that i first must update

nanoperry, 2018-08-15:
Oke thanks that is working now but now my immersion reciever ppm is not working with betaflight it works normaly i see the colors from the sticks but the fields are empty now
In configuration it say ppm

CaptainYags, 2018-10-08:
Did you get it working? What target did you use?

nanoperry, 2018-10-10:
no ppm still not working it is taking a long time before the fixit