iNavFlight/inav #3747: Kakute F4 V2 Can't arming

PTaki, 2018-08-15:
Kakute F4 V2 Can't arming
INav 2.0.0.RC6
In RC5 no problem.

In configurator, pre arming chekcs all item is green, but not arming.

Diff:
# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# feature
feature -TX_PROF_SEL
feature VBAT
feature GPS
feature CURRENT_METER
feature RSSI_ADC
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 0 32 57600 38400 57600 115200
serial 3 2 115200 57600 0 115200
serial 5 4096 115200 38400 0 115200

# led
led 0 5,2::CNIW:5
led 1 5,1::LI:6
led 2 5,0::LI:4
led 3 6,0::G:4
led 4 7,0::A:4
led 5 8,0::A:4
led 6 9,0::G:4
led 7 10,0::LI:4
led 8 10,1::LI:6
led 9 10,2::CNIW:6

# color

# mode_color
mode_color 6 0 2
mode_color 6 1 6
mode_color 6 6 10

# aux
aux 0 0 1 1300 2075
aux 1 1 0 900 1300
aux 2 2 0 1300 1700
aux 3 20 0 1700 2100
aux 4 3 3 1300 1700
aux 5 9 3 1700 2100
aux 6 8 2 1500 2100
aux 7 11 6 1600 2100
aux 8 12 1 1700 2100
aux 9 15 5 1500 2100

# adjrange

# rxrange

# osd_layout

# master
set gyro_sync = ON
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 90
set gyro_notch1_hz = 200
set gyro_notch1_cutoff = 100
set acc_hardware = MPU6500
set acczero_x = -15
set acczero_y = 13
set acczero_z = 9
set accgain_x = 4101
set accgain_y = 4099
set accgain_z = 4051
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set mag_declination = 450
set magzero_x = 17
set magzero_y = -68
set magzero_z = 246
set baro_hardware = BMP280
set pitot_hardware = NONE
set blackbox_rate_denom = 3
set blackbox_device = SPIFLASH
set min_throttle = 1160
set motor_pwm_rate = 1000
set motor_pwm_protocol = ONESHOT125
set align_board_roll = -34
set align_board_pitch = -12
set current_meter_scale = 260
set bat_voltage_src = SAG_COMP
set model_preview_type = 3
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set frsky_vfas_precision = 1
set smartport_uart_unidir = ON
set name = Kakute V2

# profile
profile 1

set mc_p_pitch = 58
set mc_i_pitch = 50
set mc_d_pitch = 22
set mc_p_roll = 43
set mc_i_roll = 40
set mc_d_roll = 20
set mc_p_yaw = 70
set dterm_lpf_hz = 80
set dterm_notch_hz = 200
set dterm_notch_cutoff = 100
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 45

# battery_profile
battery_profile 1

set vbat_min_cell_voltage = 325
set battery_capacity = 2200
set battery_capacity_warning = 506
set battery_capacity_critical = 374

# 
When the LED is enabled, Battery monitor does not work.

digitalentity, 2018-08-15:
Go to CLI, execute "status", post output.

PTaki, 2018-08-15:
# status
System Uptime: 12 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 12.0V (3S battery - OK)
CPU Clock=168MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280, MAG=HMC5883
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
Stack size: 6144, Stack address: 0x10010000, Heap available: 976
I2C Errors: 0, config size: 4217, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = ADC 3
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 1, cycle time: 1002, PID rate: 998, RX rate: 49, System rate: 9
Arming disabled flags: NAV RX CLI

#

digitalentity, 2018-08-15:
`Arming disabled flags: NAV RX CLI`

Navigation not safe, No receiver signal

PTaki, 2018-08-15:
When the transmitter is on, the result is same.

digitalentity, 2018-08-15:
What is the shown in warnings in OSD?

PTaki, 2018-08-15:
# status
System Uptime: 36 seconds
Current Time: 2018-08-15T17:42:34.947+00:00
Voltage: 11.86V (3S battery - OK)
CPU Clock=168MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280, MAG=HMC5883
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
Stack size: 6144, Stack address: 0x10010000, Heap available: 976
I2C Errors: 0, config size: 4217, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = ADC 3
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 1, cycle time: 1003, PID rate: 997, RX rate: 110, System rate: 10
Arming disabled flags: CLI

PTaki, 2018-08-15:
I have to go to work. Tomorow will be check agin.

 

Peter