iNavFlight/inav #3748: Status of compatibility with Flytron 10Hz GPS module V2 GPS

VoicOfReason, 2018-08-15:
I apologize if this should be posted elsewhere, but since it may not be compatible, I wanted to see how feasible it would be to implement compatibility.

Will this GPS work with iNav PNP?

http://www.flytron.com/osd-headtrackers/15-simpleosd-gps-module.html

teckel12, 2018-08-15:
@VoicOfReason Probably no way to know if it doesn't say what GPS module it's using. It's probably BLOX, but that's a guess. And if it is, it may not be programmed correctly for INAV, so you may need to program it correctly first.

stronnag, 2018-08-15:
"Standart MTK Commands for settings" --- whatever a Standart is. I tend not to buy stuff like that. But we dont' support MTK well, it at all.

closed 2018-08-16 by VoicOfReason