iNavFlight/inav #3749: `OMNIBUSF4V3` firmware not registering stick movements

kerrycorcoran, 2018-08-15:
### Board and Version
**FC** (https://www.banggood.com/Eachine-Wizard-X220S-FPV-Racer-Customized-Version-Omnibus-F4-V3-Betaflight-3_0_1-Flight-Controller-p-1177121.html?rmmds=myorder&cur_warehouse=CN)
**Version** 1.91 of INAV 

### Behavior
I am not getting any stick movement within iNav.  I have confirmed the TX/RX are bound (green LED), have power to the RX.  I have Wired RX SBUS lead to SBUS header on Flight Controller, and 5v and GND accordingly.
I have configured UART1 as SerialRX; disabled MSP.
Under Configuration, I have selected SerialRX and SBUS RX.

The RX had been verified as good as I can set it up on another FC (Omnibus F4V2.1) and it works without issue.  I have also tested with a FrSky X8R and get the same results; no stick activity when moved.

I have tested with another F4V3 FC and get the same results.

Reviewing other posts I have soldered the SBUS pads on the FC; same results.

This is a very similar setup I have used on several other F4 FCs without issue.

Using the following:
**RX** (https://www.banggood.com/FrSky-XSR-2_4GHz-16CH-ACCST-Receiver-S-Bus-CPPM-Output-Support-X9D-X9E-X9DP-X12S-X-Series-p-1031481.html?rmmds=myorder&cur_warehouse=CN)
**Transmitter** (https://www.banggood.com/Original-FrSky-2_4G-16CH-ACCST-Taranis-X9D-Plus-Transmitter-Carton-Package-p-1121799.html?rmmds=search&cur_warehouse=CN)

Is there something different about this Flight Controller?

Tygrys-1, 2018-08-16:
Did you have SBUS jumper connected?

dasrue, 2018-08-16:
When using betaflight, this board needs the OMNIBUSF4**SD** firmware, otherwise the SBUS input doesn't seem to work.

kerrycorcoran, 2018-08-16:
@Tygrys-1 , yes, soldered the SBUS jumper.
@dasrue , just re-flashed with a target of OMNIBUSF4SD and still no movement.

Tygrys-1, 2018-08-16:
Try other UARTS as the serial. Some omnibus implementations has SBUS on UART6

kerrycorcoran, 2018-08-16:
@Tygrys-1 , if I want to try UART6, do I wire the RX SBUS lead to the Flight Controller 6RX or 6TX pin?

teckel12, 2018-08-17:
@kerrycorcoran RX, because your receiving SBUS from your receiver. Doesn't matter what UART it is.

kerrycorcoran, 2018-08-17:
Okay, got it working!

**Target**: OMNIBUSF4V3
**Version**: 1.9.1 - OMNIBUSF4PRO
**Port** = UART6
**Protocol**: Serial Rx / SBUS

**Wiring**: RX SBUS lead to Flight Controller RX6 pin.

Do **NOT** need to solder the SBUS pad on the flight controller

Thank you to all that helped get this resolved.

closed 2018-08-17 by kerrycorcoran