iNavFlight/inav #3750: Seriously DODO Telemetry

mrduncan626, 2018-08-15:
I have a Seriously DODO F3 in which the telemetry works great with BetaFlight but after flashing this FC with iNav 2.0.0 RC6, there is no telemetry!!! Same parameters in both firmwares:

PORTS:
UART2:  SPort.
UART3:  SerialRX.

CONFIGURATION:
Serial Based RX with SBUS.
Telemetry Output enabled

CLI:
serialrx_inverted = OFF
telemetry_inverted = OFF

The Seriously DODO iNav DOC just refers you to the SPRacing FC DOC which indicates it supports telemetry but the board layouts are totally different.  

However, I have a SPRacing FC with iNav which supports telemetry perfectly.

teckel12, 2018-08-16:
@mrduncan626 If you're running SBUS and S.Port telemetry, why do you have inversion off?  Those are both inverted protocols.

mrduncan626, 2018-08-16:
The new inversion protocol is just the opposite from what you think.  These 2 inverted protocols (off) work in BetaFlight (SBUS and SPORT) but only the SBUS works in iNav.  Setting telemetry_inverted to OFF or ON doesn't work in iNav on the DODO F3.

teckel12, 2018-08-16:
@mrduncan626 No idea how betaflight does it. But with INAV, SBUS and S.Port are inverted and that's what you should select if you want inverted. If you're using F.Port, it's uninverted, or if you've wired to your RX on the uninverted pads on your Rx it's uninverted. With the F3 you can select inversion as the MCU does the inversion.

I'd forget about your BF settings if I were you. They mean nothing when using INAV. Configure from scratch, and inversion is inversion with INAV, and SBUS and S.Port are inverted protocols be default.

closed 2018-10-29 by stale[bot]