iNavFlight/inav #3753: 2.0.0-RC6 SmartAudio issues

digitalentity, 2018-08-16:
A metabug to track 2.0.0-RC6 SmartAudio issues.
Related: #3752 #3681 #3734

digitalentity, 2018-08-16:
Differences in `SA_CMD_GET_SETTINGS` command across versions

1.9.1
`baud 4800 : 0x00 0xAA 0x55 0x03 0x00 0x9F 0x00`

2.0 RC3
`baud 4800 : 0x00 0xAA 0x55 0x03 0x00 0x9F`

dev:
`baud 4800 : 0x00 0xAA 0x55 0x03 0x00 0x9F 0x00`

digitalentity, 2018-08-16:
This doesn't look right to me:

1.9.1 & 2.0.0-RC3
```
        smartAudioSerialPort = openSerialPort(portConfig->identifier, FUNCTION_VTX_SMARTAUDIO, NULL, NULL, 4800, MODE_RXTX, SERIAL_BIDIR|SERIAL_BIDIR_PP);
```

dev:
```
        portOptions_t portOptions = SERIAL_STOPBITS_2 | SERIAL_BIDIR_NOPULL;
#if defined(USE_VTX_COMMON)
        portOptions = portOptions | (vtxConfig()->halfDuplex ? SERIAL_BIDIR | SERIAL_BIDIR_PP : SERIAL_UNIDIR);
#else
        portOptions = SERIAL_BIDIR;
#endif

        smartAudioSerialPort = openSerialPort(portConfig->identifier, FUNCTION_VTX_SMARTAUDIO, NULL, NULL, 4800, MODE_RXTX, portOptions);
```

Changes: 2 stop bits in 2.0-RC6 compared to 2.0-RC3/1.9.1

It seems that in 2.0.0-RC6 responses from VTX are ignored. Data being sent with 2 stop bits will be correctly received if receiver is expecting 2 stop bits, but vice-versa it won't work.

digitalentity, 2018-08-16:
Firmware builds with attempted fix. @Sil20 @haksarion @lucaslabs @clemtrinc please test.

[inav_2.0.0_FF_F35_LIGHTNING.hex.zip](https://github.com/iNavFlight/inav/files/2294743/inav_2.0.0_FF_F35_LIGHTNING.hex.zip)
[inav_2.0.0_MATEKF405OSD.hex.zip](https://github.com/iNavFlight/inav/files/2294744/inav_2.0.0_MATEKF405OSD.hex.zip)
[inav_2.0.0_OMNIBUSF4.hex.zip](https://github.com/iNavFlight/inav/files/2294745/inav_2.0.0_OMNIBUSF4.hex.zip)
[inav_2.0.0_OMNIBUSF4PRO.hex.zip](https://github.com/iNavFlight/inav/files/2294746/inav_2.0.0_OMNIBUSF4PRO.hex.zip)
[inav_2.0.0_OMNIBUSF4V3.hex.zip](https://github.com/iNavFlight/inav/files/2294747/inav_2.0.0_OMNIBUSF4V3.hex.zip)

lucaslabs, 2018-08-16:
hey @digitalentity just tested on my FF F35 Lighting FC and this fix *is not working*. I can only see stats and back option once I entered on VTx Smart Audio feature on OSD 😞

digitalentity, 2018-08-16:
@lucaslabs from which version did you upgrade? Retry with full erase to make sure settings are consistent.

lucaslabs, 2018-08-16:
@digitalentity i've downloaded "inav_2.0.0_FF_F35_LIGHTNING.hex.zip" and flashed it using full erase chip. Smart Audio is on UART4.

clemtrinc, 2018-08-16:
@digitalentity ok will try in an hour.

clemtrinc, 2018-08-16:
@digitalentity Thanks for your help, I flashed your attempted fix firmware and it did solve my issue, smartaudio now sets the correct power at startup and arming.

Just a side note: On my first connection with your fix software, I had a GPS fail, and my gps is now (probably hardware) failed as I get 0 messages from it even when reverting to RC6 or 1.9.1
I can't care less as it's cheap gps and it's been overheating for the last 6 months so it could just be a coincidence, but I just wanted to let you know it happened immediately after flashing.

Clement

digitalentity, 2018-08-17:
Confirmed that SmartAudio works with two different boards and VTXes. Closing.

closed 2018-08-17 by digitalentity

sq9fk, 2018-09-07:
SmartAudio works on inav 2.0.0 on SoftSerials with OMNIBUSF4PRO form BG/Ali
I've tested this with VTX with `0x00` bug and without.

To configure SoftSerials with OMNIBUSF4PRO please look at my comment here:
https://github.com/iNavFlight/inav/issues/3574#issuecomment-419371066
or here:
https://github.com/iNavFlight/inav/issues/3610#issuecomment-419416745

Amenotepia, 2018-09-09:
I keep having issues with smartaudio not working properly on FF F35 Lightning, stable release, UART4. What am I missing here?