iNavFlight/inav #3761: Bug iNav 2.0.0 RC6 Failsafe disarm instead of RTH

konraddi, 2018-08-17:
As mentioned here: https://github.com/iNavFlight/inav/pull/3482

My plane disarmed because of multiple failsafes some 10km away. In previous iNav versions it never happened! Failsafe procedure is set to RTH, and the Crossfire Micro RX v2 is set to "Cut" on failsafe, no failsafe positions. Someting seems to work wrong with the latest changes...

[Dump AR Wing v6 iNav 2.0.0 RC6.txt](https://github.com/iNavFlight/inav/files/2298509/Dump.AR.Wing.v6.iNav.2.0.0.RC6.txt)

[Diff AR Wing v6 iNav 2.0.0 RC6.txt](https://github.com/iNavFlight/inav/files/2298496/Diff.AR.Wing.v6.iNav.2.0.0.RC6.txt)

[LOGS AR Wing.zip](https://github.com/iNavFlight/inav/files/2298498/LOGS.AR.Wing.zip)

Thanks,
Konrad

stronnag, 2018-08-17:
From an accelerated BB replay ... 
```
2018-08-17T20:52:45+0100 Set home 48.062918 11.853131 (LTM OFrame true)
2018-08-17T20:52:49+0100 New LTM Mode Cruise (18) 1 44s 48.063717 11.848551 0 20
2018-08-17T20:52:58+0100 New LTM Mode Return Home (13) 1 144s 48.060127 11.838222 0 4
2018-08-17T20:52:59+0100 New LTM Mode Cruise (18) 1 154s 48.060679 11.839292 0 20
2018-08-17T20:54:39+0100 Failsafe asserted
2018-08-17T20:54:39+0100 New LTM Mode Return Home (13) 1 1141s 48.087950 11.721537 0 4
2018-08-17T20:54:39+0100 Failsafe cleared
2018-08-17T20:54:39+0100 New LTM Mode Cruise (18) 1 1144s 48.088251 11.721416 0 20
2018-08-17T20:54:41+0100 Failsafe asserted
2018-08-17T20:54:41+0100 New LTM Mode Return Home (13) 1 1167s 48.087200 11.723428 0 4
2018-08-17T20:54:41+0100 Failsafe cleared
2018-08-17T20:54:41+0100 New LTM Mode Cruise (18) 1 1168s 48.087088 11.723387 0 20
2018-08-17T20:54:41+0100 Failsafe asserted
2018-08-17T20:54:41+0100 New LTM Mode Return Home (13) 1 1172s 48.086533 11.723602 0 4
2018-08-17T20:54:43+0100 Failsafe cleared
2018-08-17T20:54:43+0100 Failsafe asserted
2018-08-17T20:54:43+0100 Failsafe cleared
2018-08-17T20:54:43+0100 New LTM Mode Cruise (18) 1 1193s 48.085383 11.728754 0 20
2018-08-17T20:54:43+0100 Failsafe asserted
2018-08-17T20:54:45+0100 Failsafe cleared
2018-08-17T20:54:45+0100 Failsafe asserted
2018-08-17T20:54:46+0100 Failsafe cleared
2018-08-17T20:54:47+0100 Failsafe asserted
2018-08-17T20:54:48+0100 New LTM Mode Return Home (13) 1 1241s 48.082961 11.739466 0 4
2018-08-17T20:54:49+0100 Assumed disarm from LTM
2018-08-17T20:54:49+0100 Distance = 12946.1, max speed = 22.0 time = 1256
2018-08-17T20:54:50+0100 BB Disarm Failsafe (6)
```
Purple track : Cruise
Yellow Track: RTH

[image]

Long walk of shame ....

konraddi, 2018-08-17:
haha :D Not quite right @stronnag, look at the other files, luckily I was able to rearm after some very long time and with a lot of height loss. Rearming was probably not possible because the launch mode also interfered? Launch mode is on my channel at 1400-1600us and it seems that this channel froze at 1500us for a long time? Arming seems impossible as long as the launch mode is activated inflight!!

konraddi, 2018-08-17:
There is some option to keep the height from the start point (inav_reset_altitude | FIRST_ARM)

The same would be very helpful to keep the first Home position and not to set it to the current position after a (failsafe) disarm but keep the first GPS position where the plane was launched...

stronnag, 2018-08-17:
I saw the others logs _after_ I posted the image. Still the interesting bit is the disarm reason (6) - failsafe.

konraddi, 2018-08-17:
yep it should not disarm when in Crossfire RX failsafe (CRSF v2 with "cut") but do the RTH as normally in iNav 1.8.0. btw it was a crossfire interference test with this plane, wanted to know if some components limit the range, the TX was set to 10mw max...

digitalentity, 2018-08-17:
`LOG00001.TXT` - disarmed by failsafe (RTH failed to activate
`LOG00002.TXT` - disarmed by switch
`LOG00003.TXT` - disarmed by switch
`LOG00004.TXT` - disarmed by switch

Found a bug in `CRUISE` mode implementation. If failsafe is triggered while pilot (or frozen receiver channels) are "adjusting" direction - RTH will fail to activate and plane will fallback to `LANDING` procedure. Another bug is that even although failsafe `LANDING` should spiral the plane down - CRUISE controller seems to be active and steering the plane.

digitalentity, 2018-08-17:
Working on a fix

konraddi, 2018-08-17:
@digitalentity Thanks! Other than that, the Cruise mode in combination with ALTHOLD works superb even on a bank and yank flying wing!

@stronnag Thanks too! "inav_reset_home" is not listed (yet) in the CLI doc? (https://github.com/iNavFlight/inav/blob/master/docs/Cli.md)

digitalentity, 2018-08-17:
@stronnag interesting finding, but unrelated. There is a bug in the code that prevents plane from going into RTH if a certain set of conditions are met:

* Cruise mode activated
* RPY sticks are not centered
* Receiver goes into failsafe

digitalentity, 2018-08-17:
Fix ready for review 😄

digitalentity, 2018-08-17:
@konraddi @stronnag thanks

stronnag, 2018-08-17:
@konraddi Both reset options are documented in the development branch Cli document.

digitalentity, 2018-08-18:
Fix merged. Closing.

closed 2018-08-18 by digitalentity