iNavFlight/inav #3764: Matek F405 Wing motor doesn't arm

jeepee16, 2018-08-19:
Hello,

Setup as follows;
R9 rx running latest firmware
Matek F405 wing FC running latest firmware (INav 2.0.0. RC6)

I ordered a Matek F405 wing to upgrade my NanoTalon with it. But now I'm struggeling to configure the board properly. When the board is connected to iNav, everything is working well (servos are working, motor is running with the iNav Master switch, Pre-arming check all green, GPS working, OLED display working, VTX working, LED working, buzzer working, … everything fine) but when I disconnect the board from my computer and try to arm it, it’s beeping non stop (with the blue led flashing) and the motor doesn’t want to run (servos and everything are working). I have flashed the F405 board with the latest firmware (inav_2.0.0_rc6) but it doesn’t resolve the problem. I don’t know what is wrong, does any body have an idea ?

shellixyz, 2018-08-19:
Please go into the CLI, run the `status` command and post the result

jeepee16, 2018-08-20:
Hello,

Today, I tried to connect the flight controller to iNav and it didn't recognize it anymore (iNav message said: "Failed to open serial port"). This is so weird because yesterday it was connecting without any problem. Since yesterday, I haven't touched the flight controller and it is the same computer I used to connect before... I don't know what's wrong!

jreis1, 2018-08-23:
I've been running all inav 2.0RX with the same hardware withouth any issue.

jeepee16, 2018-08-26:
Hello,
This weekend I tried to connect the F405 board on the Macbook of my girlfriend and it worked. I also tried to connect another ESC and motor to the board because I suspected the Nano Talon ESC was malfunctioning. But I have the same issue with this other ESC/motor. In any case, I was able to retrieve the board status and diff. See it below:
# status
System Uptime: 280 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.0V (1S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=MPU6000, ACC=MPU6000
STM32 system clocks:
SYSCLK = 168 MHz
HCLK = 168 MHz
PCLK1 = 42 MHz
PCLK2 = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=NONE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
SD card: Startup failed
Stack size: 6144, Stack address: 0x10010000, Heap available: 892
I2C Errors: 0, config size: 4233, max available config: 131072
ADC channel usage:
BATTERY : configured = ADC 1, used = ADC 1
RSSI : configured = ADC 3, used = none
CURRENT : configured = ADC 2, used = ADC 2
AIRSPEED : configured = none, used = none
System load: 1, cycle time: 1002, PID rate: 998, RX rate: 148, System rate: 9
Arming disabled flags: NAV CLI

# 
DIFF Command
# diff

# version
# INAV/MATEKF405SE 2.0.0 Aug 20 2018 / 19:22:34 (dbdd1656a)

# resources

# mixer
mmix 0 1.000 0.000 0.000 0.000

# servo mix
smix 0 2 0 -100 0
smix 1 3 1 85 0
smix 2 4 1 -85 0
smix 3 3 2 75 0
smix 4 4 2 75 0

# servo

# feature
feature MOTOR_STOP
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial
serial 2 4096 115200 38400 0 115200
serial 3 2 115200 115200 0 115200
serial 4 1024 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 1700
aux 1 2 1 1300 2100
aux 2 10 1 900 1300
aux 3 18 2 1300 2100

# adjrange

# rxrange

# osd_layout

# master
set acc_hardware = MPU6000
set acczero_x = 27
set acczero_y = 12
set acczero_z = -294
set accgain_x = 4056
set accgain_y = 4109
set accgain_z = 4056
set mag_hardware = NONE
set baro_hardware = NONE
set pitot_hardware = NONE
set failsafe_procedure = RTH
set align_board_yaw = 900
set model_preview_type = 30
set small_angle = 180
set nav_rth_allow_landing = NEVER
set nav_rth_altitude = 7000
set nav_fw_bank_angle = 35

# profile
profile 1

set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 600

# battery_profile
battery_profile 1

#

stronnag, 2018-08-26:
GPS=NONE
Failsafe RTH

How's that meant to work?

jeepee16, 2018-08-26:
I removed all the components from the board (VTX, GPS, buzzer, LED) except the ESC/motor and the receiver to be sure nothing will interfere with the ESC/motor. So I set GPS=NONE

jeepee16, 2018-08-26:
@stronnag I removed the RTH failsafe and it is working now. So I guess it was a conflit with the failsafe settings. Thank you for your help.

digitalentity, 2018-08-26:
@jeepee16 there is a safeguard that prevents arming if RTH may be requested (RTH flight mode or RTH failsafe), but home position is not known. Apparently without GPS home position will never be acquired and thus "Navigation not safe" arming blocker will never clear.

closed 2018-08-26 by digitalentity