iNavFlight/inav #3769: incorrect GPS altitude on OSD > 500 meters

rozhkovets, 2018-08-20:
## Current Behavior
https://youtu.be/sNggjVCiKOY

## Steps to Reproduce
1. inav 2.0.0 RC4
2. fly and climb to the altitude ~>512

## Expected behavior
With increasing altitude the gps altitude value on osd becomes incorrect, decreases to negative values or disappears completely.
With decrease in altitude after a while everything comes to normal.

## Additional context
dump https://pastebin.com/vTjAPhH9

I can not argue that the problem is only with the OSD, because the black box was disabled

---
# INAV/MATEKF405OSD 2.0.0 Jul 14 2018 / 22:37:57 (f2cf315ad)

rozhkovets, 2018-08-21:
I updated the firmware to 2.0.0 with a complete chip erasing. Problem still exists. Attached a new video (2:53), blackbox log and google eath kmz.
video - https://youtu.be/BRm9ZRwkwMs
blackbox log and kmz - https://drive.google.com/drive/folders/1RSj63lUuZ7BKwiF2JgK7yOaFoTz_aVxs?usp=sharing

digitalentity, 2018-08-21:
Looks like this is specific to `set gps_provider = NMEA`

``
201612284, 2, 12, 58.4731000, 56.4704116, 652, 16.20, 89.0, 50, 100, 100
201693242, 2, 12, 58.4731000, 56.4704116, 652, 16.20, 89.0, 50, 100, 100
201812421, 2, 12, 58.4731000, 56.4704666, 654, 15.27, 93.9, 50, 100, 100
202012568, 2, 12, 58.4730966, 56.4705166, 1, 14.50, 98.6, 50, 100, 100
202212821, 2, 12, 58.4730916, 56.4705650, 3, 13.78, 103.7, 50, 100, 100
202412958, 2, 12, 58.4730850, 56.4706083, 5, 13.16, 109.4, 50, 100, 100
202613149, 2, 12, 58.4730766, 56.4706500, 6, 12.65, 114.8, 50, 100, 100
202813313, 2, 12, 58.4730666, 56.4706883, 8, 12.13, 122.0, 50, 100, 100
``

After 654m absolute altitude GPS provider started reporting incorrect altitude. 654m = 65400cm which gives us a hint about 16-bit overflow in either GPS itself of NMEA decoder.

digitalentity, 2018-08-21:
```
typedef struct gpsDataNmea_s {
    bool fix;
    int32_t latitude;
    int32_t longitude;
    uint8_t numSat;
    uint16_t altitude;
    uint16_t speed;
    uint16_t ground_course;
    uint16_t hdop;
    uint32_t time;
    uint32_t date;
} gpsDataNmea_t;
```

Got it. Looks like nobody flew NMEA higher than 655.35m MSL altitude 🤣

stronnag, 2018-08-21:
Or lower than MSL ...

digitalentity, 2018-08-21:
Yes, NMEA driver doesn't even consider negative altitudes

digitalentity, 2018-08-21:
#3778

fiam, 2019-01-06:
Fixed in #3778

closed 2019-01-06 by fiam