iNavFlight/inav #3771: How to use MSP_RC and MSP_SET_RAW_RC?

09Ria09, 2018-08-20:
I have this project which consists of a SPRACINGF4EVO and a Raspberry Pi Zero, but I have a few problems with MSP_SET_RAW_RC and MSP_RC. My first problem is that when using MSP_SET_RAW_RC the drone doesn't do anything and the second one is that when I send MSP_RC I get values(first 8 values: `2250 2250 2250 2250 0 0 0 0`) which aren't between 1000 and 2000.
Here is my code in Python:

    def setRawRC(self, roll, pitch, yaw, thrust):
        self.sendMessage(self.MSP_SET_RAW_RC,
                         # [roll, pitch, thrust, yaw, 0, 0, 0, 0]
                         [1500, 1500, 1000, 2000, 1000, 1000, 1000, 1000],
                         [2, 2, 2, 2, 2, 2, 2, 2])
        self.ser.timeout = 0.03
        print(struct.unpack('<3sBBB', self.ser.read(6)))
        self.ser.timeout = None
        self.sendMessage(self.MSP_RC, [], [])
        self.ser.read(3)
        databytes = struct.unpack('<B', self.ser.read())[0]
        self.ser.read(1)
        d1 = struct.unpack('<h', self.ser.read(2))[0]
        d2 = struct.unpack('<h', self.ser.read(2))[0]
        d3 = struct.unpack('<h', self.ser.read(2))[0]
        d4 = struct.unpack('<h', self.ser.read(2))[0]
        d5 = struct.unpack('<h', self.ser.read(2))[0]
        d6 = struct.unpack('<h', self.ser.read(2))[0]
        d7 = struct.unpack('<h', self.ser.read(2))[0]
        d8 = struct.unpack('<h', self.ser.read(2))[0]
        self.ser.read(21)
        print(str(databytes) + ' ' + str(d1) + ' ' + str(d2) + ' ' + str(d3) + ' ' + str(d4) + ' ' + str(d5) + ' ' +
              str(d6) + ' ' + str(d7) + ' ' + str(d8))

And here is `sendMessage`:

    def sendMessage(self, message_type, data, dnob):  # dnob - data: number of bytes
        log('   sending message to FC with message type: ' + str(message_type))
        # making the output
        length = len(data)
        output = ctypes.create_string_buffer(512)

        # Documentation: https://docs.python.org/3/library/struct.html
        struct.pack_into('<3sBB', output, 0, self.header, length, message_type)

        for i in range(0, length):
            if dnob[i] == 1:
                struct.pack_into('<B', output, 5 + i, data[i])
            elif dnob[i] == 2:
                struct.pack_into('<h', output, 5 + i, data[i])
            elif dnob[i] == 4:
                struct.pack_into('<i', output, 5 + i, data[i])

        # sending the data
        for i in range(0, 5 + length):
            self.ser.write(output[i])
        self.ser.write(struct.pack('<B', xor(output, 3, 5 + length)))

Here is what I get when I execute `status` in the CLI:

System Uptime: 22 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.8V (1S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
SD card: None inserted
Stack size: 6144, Stack address: 0x10010000, Heap available: 936
I2C Errors: 0, config size: 3509, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
[... 576 lines cut ...]
set tpa_breakpoint = 1500
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

giacomo892, 2018-08-20:
You need to change your serialrx provider in order to MSP RC to work.

09Ria09, 2018-08-20:
I did: 
[image]
And on the ports tab, each time I try to change USB VCP Serial RX to ON it changes into this state(note that UART1 MSP was turned to OFF and UART2 Serial RX was turned to OFF):
[image]

giacomo892, 2018-08-20:
why do you want to use USB port to comunicate with the RPI? Just use another uart...

09Ria09, 2018-08-20:
I use the USB port because it's easier and I already use UART on my RPI. The MSP connections works correctly but I still don't know how to make MSP_SET_RAW_RC to work, or at least MSP_RC.

shellixyz, 2018-08-20:
@09Ria09 I stated previously by @giacomo892 you only need MSP to be enabled on the port not `Serial RX` to use `MSP_SET_RAW_RC`

09Ria09, 2018-08-21:
@giacomo892 @shellixyz I changed the ports so it looks like this: 
[image] , but `MSP_SET_RAW_RC` still doesn't work, and `MSP_RC` still returns values which aren't between 1000 and 2000(first 8 are: `2250 2250 2250 2250 0 0 0 0`)

09Ria09, 2018-08-21:
any ideas?

09Ria09, 2018-08-23:
I will try to see if this works with cleanflight or betaflight.

09Ria09, 2018-08-25:
MSP_RC work correctly in Cleanflight and Betaflight, but MSP_SET_RAW_RC still doesn't work. I already set receiver mode to MSP RX Input. What is wrong? I've changed my code so I can test it using my PC. Here it is:
```
from serial import Serial
import struct
import ctypes

MSP_SET_RAW_RC = 200
MSP_RC = 105

header = b'$M<'
ser = Serial('COM3', baudrate=115200)

def xor(data, startingPoint, bEndPoint):
    result = 0
    for i in range(startingPoint, bEndPoint):
        result ^= ord(data[i])
    return result

def sendMessage(message_type, data, dnob):  # dnob - data: number of bytes
    print('   sending message to FC with message type: ' + str(message_type))
    # making the output
    length = len(data)
    output = ctypes.create_string_buffer(512)

    # Documentation: https://docs.python.org/3/library/struct.html
    struct.pack_into('<3sBB', output, 0, header, length, message_type)

    for i in range(0, length):
        if dnob[i] == 1:
            struct.pack_into('<B', output, 5 + i, data[i])
        elif dnob[i] == 2:
            struct.pack_into('<h', output, 5 + i, data[i])
        elif dnob[i] == 4:
            struct.pack_into('<i', output, 5 + i, data[i])

    # sending the data
    for i in range(0, 5 + length):
        ser.write(output[i])
    ser.write(struct.pack('<B', xor(output, 3, 5 + length)))

sendMessage(MSP_SET_RAW_RC,
            [1500, 1500, 1500, 1000],
            [2, 2, 2, 2])
ser.timeout = 0.03
# ser.read(6)  # skipping the response check for a lower latency

print(struct.unpack('<3sBBB', ser.read(6)))
ser.timeout = None

sendMessage(MSP_RC, [], [])
ser.read(3)
databytes = struct.unpack('<B', ser.read())[0]
ser.read(1)
d1 = struct.unpack('<h', ser.read(2))[0]
d2 = struct.unpack('<h', ser.read(2))[0]
d3 = struct.unpack('<h', ser.read(2))[0]
d4 = struct.unpack('<h', ser.read(2))[0]
d5 = struct.unpack('<h', ser.read(2))[0]
d6 = struct.unpack('<h', ser.read(2))[0]
d7 = struct.unpack('<h', ser.read(2))[0]
d8 = struct.unpack('<h', ser.read(2))[0]
ser.read(21)
print(str(databytes) + ' ' + str(d1) + ' ' + str(d2) + ' ' + str(d3) + ' ' + str(d4) + ' ' + str(d5) + ' ' +
      str(d6) + ' ' + str(d7) + ' ' + str(d8))'
```

stronnag, 2018-08-25:
Still works perfectly here, script referenced above modified to print MSP_RC results:
```
./msp_set_rc.rb -d /dev/ttyACM0 
dev /dev/ttyACM0 115200
[
    [0] 231,
    [1] 3,
    [2] 0,
    [3] 2147483708
]
Tx: 1507,1582,1564,1581,1017,1442,1663,1969
Rx: 1507,1582,1564,1581,1017,1442,1663,1969

Tx: 1533,1581,1599,1500,1017,1442,1663,1969
Rx: 1533,1581,1599,1500,1017,1442,1663,1969

Tx: 1585,1515,1527,1599,1017,1442,1663,1969
Rx: 1585,1515,1527,1599,1017,1442,1663,1969

Tx: 1537,1564,1555,1543,1017,1442,1663,1969
Rx: 1537,1564,1555,1543,1017,1442,1663,1969

Tx: 1515,1560,1551,1543,1017,1442,1663,1969
Rx: 1515,1560,1551,1543,1017,1442,1663,1969

Tx: 1570,1516,1599,1591,1017,1442,1663,1969
Rx: 1570,1516,1599,1591,1017,1442,1663,1969

Tx: 1522,1506,1540,1573,1017,1442,1663,1969
Rx: 1522,1506,1540,1573,1017,1442,1663,1969

Tx: 1552,1524,1570,1528,1017,1442,1663,1969
Rx: 1552,1524,1570,1528,1017,1442,1663,1969

Tx: 1580,1576,1513,1546,1017,1442,1663,1969
Rx: 1580,1576,1513,1546,1017,1442,1663,1969

Tx: 1535,1513,1567,1502,1017,1442,1663,1969
Rx: 1535,1513,1567,1502,1017,1442,1663,1969

Tx: 1586,1523,1559,1527,1017,1442,1663,1969
Rx: 1586,1523,1559,1527,1017,1442,1663,1969

Tx: 1560,1519,1523,1509,1017,1442,1663,1969
Rx: 1560,1519,1523,1509,1017,1442,1663,1969

Tx: 1587,1518,1516,1522,1017,1442,1663,1969
Rx: 1587,1518,1516,1522,1017,1442,1663,1969

Tx: 1505,1522,1518,1511,1017,1442,1663,1969
Rx: 1505,1522,1518,1511,1017,1442,1663,1969
```
There is no problem with iNav.

09Ria09, 2018-08-25:
I don't know ruby, so can you tell me how to set it up(what libraries to install, etc...)

stronnag, 2018-08-25:
iNav issues are for firmware bugs, not 'teach me how to programme''

09Ria09, 2018-08-25:
@stronnag you posted a script which isn't in python, and when I ask you to tell me how to use it, you respond like a douche...

09Ria09, 2018-09-05:
bump? I wonder if I would be implementing MSP V2 instead of V1, would MSP_SET_RAW_RC work?

closed 2018-11-18 by stale[bot]