iNavFlight/inav #3774: [MATEK F722 & FCHUB-W] lost control Inav 2.0 RC6

spacyje, 2018-08-20:
Hello,
I crashed with my flying wing in a very strange way since I installed a PDB Matek FCHUB-W and FC F722 STD under INAV 2.0 RC6.
The first flight went smoothly but during the second flight, I felt like a loss of control for 2-3 seconds, as if the flying wing was stuck on the last order given.I was however not far and I neither lost telemetry, RSSI ok, no failsafe.After that I landed, quickly checked the connectors and I decided to try again one last flight....Big mistake!
After taking off and after about 10 seconds, the same problem, only this time it was turning and with mid throttle. I lost control, the wing being stuck on a roll and mid throttle and crashing  to the ground. The machine was shredded... 
During the descent of death I could not do anything, the command did not answer any more, RSSI to 80, telemetry ok. I also failed to disarm, the engine continued to run on the ground for a few seconds which stopped
I come to ask your help to know if it has already happened to someone with this version of INAV because I don't know or look for the problem so far and I don't understand why everything has stuck like this.
I also send you a dump if it helps, unfortunately I don't have log.

Thank you in advance for your help.

My setup:

- ARWING 900mm
- ESC 30a
- Motor 2204 2300KV
- PDB FCHUB-W
- FC MATEK F722 STD
- VTX AKK 1200mw & Runcam sparrow
- GPS Radiolink SE 100
- RX FRSKY X4R SBUS
- Lipo 4S 2000mah
- Radio FRSKY X7

# version
# INAV/MATEKF722 2.0.0 Aug  8 2018 / 18:37:07 (1c3b41527)

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 50 0
smix 1 3 1 50 0
smix 2 4 0 -50 0
smix 3 4 1 50 0

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 -100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
[... 431 lines cut ...]
set tpa_breakpoint = 1300
set fw_tpa_time_constant = 0
set rc_expo = 40
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 1300
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

digitalentity, 2018-08-20:
Loss of control is indication of either radio failsafe or bad receiver connectivity. Can you provide blackbox log or DVR recording of the flight?

spacyje, 2018-08-20:
On the first flight a few minutes before I was over a mile away.

digitalentity, 2018-08-20:
1.2W is quite high power. It may cause interference to wires nearby and possible disrupting SBUS communications. On your next build make sure your VTX is as far as possible from FC, receiver and other wires.

spacyje, 2018-08-20:
you think? After measurements the VTX antenna is about 20cm from the RX and FC.

davidngrc, 2018-08-21:
my friend also using 2.0RC6 and lost DVR signal last sunday, the plane never went back home, I guest it hit the ocean.
I don't have his detail setup, but here is the DVR I recorded.
https://youtu.be/iUOuHG4hP04

davidngrc, 2018-08-21:
I assume this happen, when my friend hit return to home switch.
the flying wing start to turn, as shown on the fpv video feed.

then somehow the power wire to vtx got loose, that is why we are seeing a stable battery voltage at 15.8v till the end, but losing fpv video slowly due to vtx solder some capacitor.
if the vtx die or without the capacitor, we will lose fpv video immediately, not slowly.
but that does not explain why the plane never came back, because we stay there for 5 hours.

after the fpv video feed got lost, I starting to notice my friend's x9d vibrating, I assume it is the failsafe.
then 2 or 3 second pass, my friend take off his fatshark, and said he lost his plane.
we all are stunned, because he is the iNav master among us, the last plane he lost was about 4 years ago he said.

[image]

digitalentity, 2018-08-21:
After a more detailed investigation I believe this is the same issue as described in https://github.com/iNavFlight/inav/issues/3761. Fixed in 2.0 final

closed 2018-08-21 by digitalentity