iNavFlight/inav #3776: Random mag issue. Power cycling fixes it

Mckenzie74, 2018-08-21:
This happened on a 7 inch quadcopter perfectly working on inav 1.9.1
Matek F405-CTR and BN-880 gps with mag.
I waited for the gps home fix beeps with 10 sats and 1.9 hdop and took off in alt_hold as usual (I have trees around and need a precision take off), after 16 seconds I switched into pos_hold and all seemed right. Started to climb and at second 47 I turned right, during the turn the copter banks hard to the left and I disengaged the pos_hold regaining control (well, sort of, because i got disoriented and trees were blocking video signal, but alt_hold and cold blood did the rest). Looking at the DVR the heading indicator in upright corner is clearly all messed up and home arrow is indicating random places. So to me this indicates a bad magnetometer, but this quad is flying well since a couple months and continued to fly perfectly after this random issue. What fixed the problem was a power cycle. I shut it down, waited 5 seconds and reconnected the battery, voilà, problem gone, mag working again.
This could be the same issue as #3772 
Something similar happened in the same days and place with 2 more inav crafts, those are fixed wings with matek 405-wing and 411-wing flying without a magnetometer, only gps. I have logs and dvrs showing the issue also for one of those. They are on 2.0RC6. Let's see how this issue will evolve and I will eventually open a new one for the wings.
Attached are a link to dvr, bblog, dump. 
Thank you all.
Cristian.

dvr -> https://youtu.be/cQlfofvEETU
bblog -> 
[7inch a ramengo BBLOG.zip](https://github.com/iNavFlight/inav/files/2305439/7inch.a.ramengo.BBLOG.zip)
dump->
[7 inch a ramengo DUMP.txt](https://github.com/iNavFlight/inav/files/2305445/7.inch.a.ramengo.DUMP.txt)

Mckenzie74, 2018-08-21:
Thanks for your time stronnag. Going to change the gps module

Mckenzie74, 2018-08-21:
Just a thought: mwp's compass anomaly detector seems something good.... it would be awesome to have it implemented in inav firmware, isn't it? I'm no one in coding, maybe I just said something silly!

stronnag, 2018-08-22:
No really, it's naive, it does not take into account any IMU stuff; it's as trivial as:
* Speed > 3m/s
* Suitable flight mode
* GPS-Mag deviation > 45° for > 3s (most parameters configurable).