iNavFlight/inav #3777: SPracing F3 crashes with INAV 2.0

skorokithakis, 2018-08-21:
I just flashed 2.0 onto my Eachine Wizard x220 and tried to perform some flips in horizon mode with airmode on. As I did some flips, on the last one the quad started to lose altitude and fall in a spiral. Luckily, it fell on some bushes and didn't break.

I started it up again and tried to see what happened, did a flip and, as it returned to level, it started to spiral out of control again, but this time with more throttle and jumped over my head onto the building behind me.

This never happened with Betaflight, and this is the first time I tried to fly with INAV.

giacomo892, 2018-08-22:
Have you missed the issue template perhaps?

skorokithakis, 2018-08-22:
Hmm, where is the template? One didn't appear when filing the issue.

skorokithakis, 2018-08-22:
[image]

skorokithakis, 2018-08-22:
Updated, thanks.

giacomo892, 2018-08-22:
Check if ESC are calibrated. BF and INAV uses different ranges.

skorokithakis, 2018-08-22:
Done, thanks. This might be a PID tuning/parameter issue, but I haven't found a way to restore my good Betaflight config and I don't know enough to configure it from scratch.

skorokithakis, 2018-08-22:
I would, but I don't have much flying room now and I narrowly avoided breaking the quad the first time around... Maybe the PID values were inappropriate for my quad.

pDR8iLQ4go, 2018-11-01:
Maybe I have the same problem with a SP Racing F3 and INAV 2.0.0. My copter crashed two times, I do not know why. Using the previous INAV Version was fine for some months.