iNavFlight/inav #3780: Remove unnecessary null pointer checks

elfring, 2018-08-22:
[An extra null pointer check is not needed in functions](https://stackoverflow.com/questions/18775608/free-a-null-pointer-anyway-or-check-first "Free a null pointer anyway or check first?") like the following.
- [USBD_AUDIO_DeInit](https://github.com/iNavFlight/inav/blob/dbdd1656a84ee4f09ebc14b4c0b86a9476a48f46/lib/main/STM32F4/Middlewares/ST/STM32_USB_Device_Library/Class/AUDIO/Src/usbd_audio.c#L395 "USBD_AUDIO_DeInit function")
- [USBD_HID_DeInit](https://github.com/iNavFlight/inav/blob/dbdd1656a84ee4f09ebc14b4c0b86a9476a48f46/lib/main/STM32F7/Middlewares/ST/STM32_USB_Device_Library/Class/HID/Src/usbd_hid.c#L282 "USBD_HID_DeInit function")

digitalentity, 2018-08-22:
Removing NULL pointer check there creates a potential NULL-pointer dereference bug. I strongly advise against removing there kinds of checks.

digitalentity, 2018-08-22:
Moreover, those functions are not used by INAV and are optimized out by GCC

closed 2018-08-22 by digitalentity

elfring, 2018-08-22:
[A POSIX compliant free() function](http://pubs.opengroup.org/onlinepubs/9699919799/functions/free.html "Free allocated memory") will return immediately if a null pointer was passed. Thus I do not see a need to repeat such a sanity check.

digitalentity, 2018-08-22:
If you would read the code you'll notice that a point is dereferenced first and then freed:

```
  /* DeInit  physical Interface components */
  if(pdev->pClassData != NULL)
  {
   ((USBD_AUDIO_ItfTypeDef *)pdev->pUserData)->DeInit(0);
    USBD_free(pdev->pClassData);
    pdev->pClassData = NULL;
  }
```

Regardless, we don't have dynamic memory allocation in INAV, these functions are not used and they are in a 3rd party library which may (and will) be updated eventually (discarding the fixes).