iNavFlight/inav #3781: Update flymode for crsf telemetry to match 2.0.0 OSD fly modes

ZinhoRacing, 2018-08-22:
In telemetry/crsf.c file, flymodes are as follow (4 modes)

const char *flightMode = "ACRO";
    if (FLIGHT_MODE(FAILSAFE_MODE)) {
        flightMode = "!FS";
    } else if (FLIGHT_MODE(ANGLE_MODE)) {
        flightMode = "ANGL";
    } else if (FLIGHT_MODE(HORIZON_MODE)) {
        flightMode = "HOR";
}

In io/osd.c file , flymodes are as follow (13 modes)
 case OSD_FLYMODE:
        {
            char *p = "ACRO";

            if (isAirmodeActive()) {
                p = "AIR ";
            }

            if (FLIGHT_MODE(FAILSAFE_MODE)) {
                p = "!FS!";
            } else if (FLIGHT_MODE(MANUAL_MODE)) {
                p = "MANU";
            } else if (FLIGHT_MODE(NAV_RTH_MODE)) {
                p = "RTH ";
            } else if (FLIGHT_MODE(NAV_POSHOLD_MODE)) {
                if (FLIGHT_MODE(NAV_ALTHOLD_MODE)) {
                    // 3D HOLD
                    p = "HOLD";
                } else {
                    p = " PH ";
                }
            } else if (FLIGHT_MODE(NAV_CRUISE_MODE) && FLIGHT_MODE(NAV_ALTHOLD_MODE)) {
                p = "3CRS";
            } else if (FLIGHT_MODE(NAV_CRUISE_MODE)) {
                p = "CRS ";
            } else if (FLIGHT_MODE(NAV_ALTHOLD_MODE) && navigationRequiresAngleMode()) {
                // If navigationRequiresAngleMode() returns false when ALTHOLD is active,
                // it means it can be combined with ANGLE, HORIZON, ACRO, etc...
                // and its display is handled by OSD_MESSAGES rather than OSD_FLYMODE.
                p = " AH ";
            } else if (FLIGHT_MODE(NAV_WP_MODE)) {
                p = " WP ";
            } else if (FLIGHT_MODE(ANGLE_MODE)) {
                p = "ANGL";
            } else if (FLIGHT_MODE(HORIZON_MODE)) {
                p = "HOR ";
}

It would be great to have the same flymodes for crsf telemetry.
And if possible it would be nice to have an additionnal flymode "DARM"  when craft is desarmed, only  for crsf telemetry (not OSD). 

I'm available to test :), I'm using a Matek F405 wing.

wx4cb, 2018-08-22:
available to test with f405 wing and furious fpv f35 - full

ZinhoRacing, 2018-08-23:
I just managed to make the modifications and compile.
Just tryed on my plane and fly modes are working as expected on my Horus x10s on Open tx 2.2.2
I don't know how it works to submit the file to the official release.

wx4cb here is the link to my modified HEX, target MATEKF405SE, if you want to have a try.
[inav_2.0.0_CRSFTelem_FlyModesUpdate_MATEKF405SE.zip](https://github.com/iNavFlight/inav/files/2316293/inav_2.0.0_CRSFTelem_FlyModesUpdate_MATEKF405SE.zip)

teckel12, 2018-08-24:
@ZinhoRacing If you want to post the changes I'll create a PR for inclusion in INAV.

ZinhoRacing, 2018-08-24:
hi @teckel12 I'm still working on it (adding some error codes to flyMode when disarmed and disabled to arm, it is a nice to have when I go flying without all the FPV gear, just radio with telemtry and my plane).
little preview of crsf.c

} else if (ARMING_FLAG(ARMING_DISABLED_ALL_FLAGS)){
		/*  Enable to Arm, provide error code following list below :
			EA01 : Invalide setting
			EA02 : Need to power-cycle
			EA03 : Turn Arm Switch OFF
			EA04 : Not receiving RX data
			EA05 : Disabled by failsafe
			EA06 : Aircraft is not leveled
			EA07 : Sensors calibration
			EA08 : System overload
			EA09 : Navigation is unsafe
			EA0A : Compass not calibrated
			EA0B : Accelerometer not calibrated
			EA0D : Gyro failure
			EA0E : Accelerometer failure
			EA0F : Compass failure
			EA10 : Barometer failure
			EA11 : GPS failure
			EA12 : Range finder failure
			EA13 : Pitot failure
			EA14 : Hardware failure
			EA15 : Disabled by failsafe
			EA16 : Disable by killswitch
			EA17 : No RC link
			EA18 : Throttle is not low
			EA19 : Roll or Pitch not centered
			EA1A : Autotrim active
			EA1B : Not enough memory
			EA1C : CLI is active
			
		*/
		unsigned invalidIndex;
		if (ARMING_FLAG(ARMING_DISABLED_INVALID_SETTING) && !settingsValidate(&invalidIndex)) {
			flightMode = "EA01";	
				
		} else {
			switch (isArmingDisabledReason()) {
				case ARMING_DISABLED_FAILSAFE_SYSTEM:
					// See handling of FAILSAFE_RX_LOSS_MONITORING in failsafe.c
					if (failsafePhase() == FAILSAFE_RX_LOSS_MONITORING) {
						if (failsafeIsReceivingRxData()) {
							if (isUsingSticksForArming()) {
								// Need to power-cycle
								//return OSD_MESSAGE_STR("POWER CYCLE TO ARM");
								flightMode = "EA02";
								break;
							}
							// If we're not using sticks, it means the ARM switch
							// hasn't been off since entering FAILSAFE_RX_LOSS_MONITORING
							// yet
							// return OSD_MESSAGE_STR("TURN ARM SWITCH OFF");
							flightMode = "EA03";
							break;
						}
						// Not receiving RX data
						// return OSD_MESSAGE_STR(RC_RX_LINK_LOST_MSG);
						flightMode = "EA04";
						break;
					}
					// return OSD_MESSAGE_STR("DISABLED BY FAILSAFE");
					flightMode = "EA05";
					break;
				case ARMING_DISABLED_NOT_LEVEL:
					// return OSD_MESSAGE_STR("AIRCRAFT IS NOT LEVEL");
					flightMode = "EA06";
					break;

I'll test during the week end. I hope.
Not sure as well if it is the cleanest way to do it ...

teckel12, 2018-08-24:
@ZinhoRacing I don't think all your code was pasted.  I get just the first parts of the arming disabled flags.

wx4cb, 2018-08-24:
i would suggest not using cryptic error codes, we all see what they do with windows hehe

What I would do is have "english" (read: easy to understand by noobs) error messages so thqat when they post in the fb group or on here they can be asked "well what does the OSD say" instead of, what's the error code.

I know for myself trying to remember such error codes, which while reduces code size, is such a PITA on the flight line and would have to resort to looking it up.

Keep working on it though. I haven't had chance to test your build yet, but looking at the code you have posted, it looks good

teckel12, 2018-08-24:
Is there a character limit for the flight mode?  My guess is that it's only 4 characters long as Angle was shortened to just "ANGL".

wx4cb, 2018-08-24:
@teckel12  that's what I would assume as well, that seems to be the reason why these (below) are in there:

				// return OSD_MESSAGE_STR("AIRCRAFT IS NOT LEVEL");

teckel12, 2018-08-24:
I think we should make this two different pull requests.  First, add the flight modes.  Then, we can have a discussion on how failure conditions should be communicated.

The reason it should probably be in a different pull request from my perspective is that it seems like we're duplicating code in multiple places.  Also, I can see failure conditions being communicated with other telemetry protocols (like S.Port and F.Port) which would duplicate the code further.  Instead, we may want to have all telemetry protocols (maybe in addition to the OSD) use the same code base.

ZinhoRacing, 2018-08-24:
Yes, that's what i understood reading the existing code in file telemetry/csrf.c

existing code is

`
/*
0x21 Flight mode text based
Payload:
char[]      Flight mode ( Null­terminated string )
*/
void crsfFrameFlightMode(sbuf_t *dst)
{
    // just do Angle for the moment as a placeholder
    // write zero for frame length, since we don't know it yet
    uint8_t *lengthPtr = sbufPtr(dst);
    sbufWriteU8(dst, 0);
    crsfSerialize8(dst, CRSF_FRAMETYPE_FLIGHT_MODE);

    // use same logic as OSD, so telemetry displays same flight text as OSD
    const char *flightMode = "ACRO";
    if (FLIGHT_MODE(FAILSAFE_MODE)) {
        flightMode = "!FS";
    } else if (FLIGHT_MODE(ANGLE_MODE)) {
        flightMode = "ANGL";
    } else if (FLIGHT_MODE(HORIZON_MODE)) {
        flightMode = "HOR";
    }
    crsfSerializeData(dst, (const uint8_t*)flightMode, strlen(flightMode));
    crsfSerialize8(dst, 0); // zero terminator for string
    // write in the length
    *lengthPtr = sbufPtr(dst) - lengthPtr;
}
`

That's why i kept 4 characters for the errors code. that's only to have an information on why we can't arm. Perhaps, can try to have more readable code for usual reasons , like Throtttle too high can be ETRH for Error Throttle...

I just tested my new version, and it is working great, errors code are well transmit, of course some of the codes are not smart ... because when there is no rx, the radio can't receive anything ... when in CLI mode, FC is not sending anything to the radio as well.
So there is some clean to make (i just copied the code from osd.c, that's the message we can see on OSD, that's the code in comment in my previews post.)

I'll try to put a message with more than 4 characters.

Here is my last code with error codes, if you have time to try.

[inav_2.0.0_CRSFTelem_FlyModesUpdate_AddEnableArmErrorCode_MATEKF405SE.zip](https://github.com/iNavFlight/inav/files/2319900/inav_2.0.0_CRSFTelem_FlyModesUpdate_AddEnableArmErrorCode_MATEKF405SE.zip)

teckel12, 2018-08-24:
@ZinhoRacing  Again, I would need your changes to include in INAV.  The target flash doesn't provide that.

Please paste the code in your reply between 3 back quotes.  The single back quote is for a single line of code.  Use the "Preview" tab to review what your post will look like to others.

ZinhoRacing, 2018-08-24:
@teckel12 , i agree with you. I tryed to make a function not to duplicate code. to share the flyMode with OSD and crsf telemetry. In fact it will  be cleaner to do it for all telemetry protocol .

I didn't posted my code yet because it was more a prouf of concept, to see if it could work.
And as i said i was not happy to duplicate code lines ...

here is my last crsf.c file.

[crsf.zip](https://github.com/iNavFlight/inav/files/2320003/crsf.zip)

teckel12, 2018-10-05:
@ZinhoRacing From another request (https://github.com/iNavFlight/LuaTelemetry/issues/36), I'm working on merging your request in this issue with another.  What target do you want to test and is it INAV 2.0?

ZinhoRacing, 2018-10-05:
Hi @teckel12 , Great.
I'm using Inav 2.0 and Matek F405 Wing. Will be happy to test your code, thank you :)

For now, I'm running my own release with the flight modes updated (including the arming status) and as well the MSP over crsf capability added. It all permit me to run a nice widget (FrskyTelemetryScript from @yaapu) on Horus X10 with OpenTx2.2.2 .
I think the modifications I did in Inav should make it possible to run your Script (LuaTelemetry) using crsf telemetry + MSP over crsf (I get all missing telemetry parameters like distance to home, home angle etc... over MSP requests, I also opened an issu on Inav : MSP over Crossfire CRSF protocol #3886)
Let me know if you need anything from me.

teckel12, 2018-10-05:
@ZinhoRacing @wx4cb Here's the build but without the error codes.  I hope it's the correct targets.  If not, let me know the exact target name (as per INAV).

[inav_2.0.0_MATEKF405SE.hex.zip](https://github.com/iNavFlight/inav/files/2450207/inav_2.0.0_MATEKF405SE.hex.zip)
[inav_2.0.0_FF_F35_LIGHTNING.hex.zip](https://github.com/iNavFlight/inav/files/2450205/inav_2.0.0_FF_F35_LIGHTNING.hex.zip)

The modes are either `RDY` (ready), `NRDY` (not ready), or one of the standard OSD flight modes.  `NRDY` would be for any of the error reasons (which you could get the reason from your goggles).

The error codes are removed because they were cryptic, the reason could be reviewed in the goggles, and it was duplicated OSD code that probably should be unified in one place if used in two places.  Not that it can't be added later after it's sorted out.

ZinhoRacing, 2018-10-05:
@teckel12 Thank you, that's the right target (MATEKF405SE). I'll try tonight on bench.
Can i have the list of the modes ? (copie of the code or the file ?) I would like to update my widget to match the modes.
Thanks.

teckel12, 2018-10-05:
@ZinhoRacing Here's the latest build:

[inav_2.0.0_MATEKF405SE.hex.zip](https://github.com/iNavFlight/inav/files/2450965/inav_2.0.0_MATEKF405SE.hex.zip)
[inav_2.0.0_FF_F35_LIGHTNING.hex.zip](https://github.com/iNavFlight/inav/files/2450967/inav_2.0.0_FF_F35_LIGHTNING.hex.zip)

After reviewing, I missed something.  That's the problem with trying to do this without testing.

ZinhoRacing, 2018-10-07:
Hi @teckel12 . Just found some time to test your code.
It is working fine.
But looks like you changed RDY to OK and NRDY to WAIT
Do you have the source code ? I could do a review if you wish.
Thank you.
Antonio

ZinhoRacing, 2018-10-07:
@teckel12 , just saw your post on Lua Telem issu 36. I posted there.

shellixyz, 2018-12-06:
Solved, changes have been merged, closing

closed 2018-12-06 by shellixyz