iNavFlight/inav #3784: Copter only throttle while start in acro mode

TIMSCHMITZ, 2018-08-23:
## Current Behavior
The copter accepts only throttle while arming at acro mode. Starting in Angle or Horizon Mode works properly.
When flying in angle mode and switching to acro there is no problem.
## Steps to Reproduce

1. Attach Battery
2. Switch to Acro Mode
3. Arm
4. Takeoff

## Expected behavior
The copter should stabilized in the air
## Suggested solution(s)

## Additional context

# version
# INAV/OMNIBUSF4V3 1.9.1 Apr 21 2018 / 13:13:58 (03a5c1922)

# resources

# mixer

# servo mix

# servo

# feature
feature -OSD
feature MOTOR_STOP
feature SOFTSERIAL
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TEAR

# name
name X520D-SR

# serial
serial 0 2 115200 57600 0 115200
serial 5 64 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 3 1500 2100
aux 1 1 0 900 1325
aux 2 2 0 1350 1750
aux 3 3 2 1275 1700
aux 4 9 2 1700 2100
aux 5 18 4 1500 2100

# adjrange

# rxrange

# master
set gyro_sync = ON
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 70
set gyro_notch1_hz = 170
set gyro_notch1_cutoff = 125
set gyro_notch2_hz = 85
set gyro_notch2_cutoff = 43
set acc_hardware = MPU6000
set acczero_x = 200
set acczero_y = -11
set acczero_z = -530
set accgain_x = 4073
set accgain_y = 4073
set accgain_z = 4056
set mag_hardware = HMC5883
set mag_declination = 190
set magzero_x = -79
set magzero_y = -169
set magzero_z = -18
set baro_hardware = BMP280
set pitot_hardware = NONE
set receiver_type = SERIAL
set serialrx_provider = SBUS
set failsafe_throttle = 1575
set failsafe_throttle_low_delay = 0
set vbat_max_cell_voltage = 420
set battery_capacity = 6000
set imu_dcm_ki = 0
set gps_sbas_mode = AUTO
set nav_extra_arming_safety = OFF
set nav_mc_hover_thr = 1575
set async_mode = GYRO

# profile
profile 1

set mc_p_pitch = 80
set mc_d_pitch = 18
set mc_p_roll = 80
set mc_d_roll = 18
set mc_p_yaw = 95
set dterm_notch_hz = 125
set dterm_notch_cutoff = 90
set pidsum_limit = 0
set rate_accel_limit_roll_pitch = 3600
set rate_accel_limit_yaw = 720
set heading_hold_rate_limit = 30
set roll_rate = 40
set pitch_rate = 40

wx4cb, 2018-08-24:
unless i'm missing something, i don't understand... acro mode is no stabilisation. At least in the normal sense of it. if you want stabilisation, then you use angle/horizon

TIMSCHMITZ, 2018-08-25:
> I believe what he's saying is that it's different when he arms in acro mode compared to when he switches from angle to acro after arming. Also, there is stabilization in acro mode (which is why STAB was renamed ANGL) here: b2a937e

Hey, what Tim Explained is exactly right but one fact is missing.
The pid controller isn’t function when arming in acro. The copter only accepts throttle roll pitch and yaw inputs via rc are ignored. And if you tilt the copter at angle mode one side of motors spins up the other slows down as I would suggest. When armed in acro mode and the copter was tilted no of the motor reacts. If you switch so angle mode and back to acro the pidcontroller works fine recognized by tilting the copter results in speeding up motors and slows down the others.

giacomo892, 2018-08-25:
You need AIRMODE enabled.

andre92, 2018-09-02:
I had a similar issue where there would be no pid output at all on low throttle and with airmode disabled. It appears having the throttle below min_check will put all motors at minthrottle or mincommand when airmode is disabled.
getMotorStatus() from mixer.c will return MOTOR_STOPPED_USER even if motor_stop is not in use

closed 2018-11-15 by stale[bot]