iNavFlight/inav #3786: Compass not recognized on OMNINXT F7

CaptainYags, 2018-08-24:
## Current Behavior
I'm using this M8N GPS compass combo
https://smile.amazon.com/gp/product/B06XR6JHDW/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1
The compass is supposedly a 5883 clone

iNav version INAV/OMNIBUSF7NXT 2.0.0 Aug 20 2018 / 19:15:39 (dbdd1656a)

The GPS works, and is connected to Tx\Rx6. The MAG is connected to SCL\SDA. The MAG is not recognized in AUTO or by manually selecting HMC5883 or QMC5883.  I have previously used this same MAG on an OMNIBUS F3 in iNav successfully.

I saw that some F7 boards need pull-up resistors, so I connected 2200 ohm resistors from the 3.3v pad on the FC to the SCL\SDA pads on the FC.  It didn't work.

Is there anything else I can try?  Could there be something missing in the firmware?

CaptainYags, 2018-08-28:
Is there some way I can test if the I2C is working on my board?  Is it supported in the code yet?

ninja-zx11, 2018-08-29:
I can confirm that genuine hmc5883 from drotek works okay with omninxtF7.It could be your hardware.

CaptainYags, 2018-08-29:
I think it means that the chip is 3.3v, but that there's a voltage regulator onboard. It's originally for an APM system, which is entirely 5v. It worked at 5v on my APM and on my F3. GPS is working on the OMNINXT, just not the compass.

If yours works, maybe I just need a different compass module for now. On a side note, how do you like the built in barometer? I heard recent rumors that it wasn't very good for alt hold

ninja-zx11, 2018-08-29:
Regarding inbuilt baro, allready created an issue here:
https://github.com/iNavFlight/inav/issues/3697

Yes it doesn't look precise but I never tried flying.I just bench tested it.May be it's okay for flying.

Just for curiosity,you can try feeding your GPS 3.3v.Did you open it up and check the 3.3v regulator?It could be a different design.After all all Fake Chinese modules with poor quality.

CaptainYags, 2018-10-09:
To wrap this up, when I unmounted my GPS for another project, I saw that the MAG chip got knocked off in a crash.  Oops!  Been trying other boards, so I can't comment on the baro

closed 2018-10-09 by CaptainYags