iNavFlight/inav #3787: Pre-arm checklist error in "Settings validated"

sebmirasol, 2018-08-24:
Hey guys!

I configured everything for a flying wing, saved and rebooted, added some stuff through CLI and saved via the command. But in the pre-arm check (inav 2.0.0) I still get an error in "Settings validated)

This is what I get from **status**
System Uptime: 614 seconds
Current Time: 2018-08-24T11:49:18.081+00:00
Voltage: 11.27V (3S battery - OK)
CPU Clock=168MHz, GYRO=MPU6500, ACC=MPU6500
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=NONE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: None inserted
Stack size: 6144, Stack address: 0x10010000, Heap available: 920
I2C Errors: 0, config size: 4135, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 1, used = ADC 1
  AIRSPEED : configured = none, used = none
System load: 1, cycle time: 1000, PID rate: 1000, RX rate: 129, System rate: 9
Arming disabled flags: CLI SETTINGFAIL
Invalid setting: report_cell_voltage

**The dum** is attached as a .txt
[dump.txt](https://github.com/iNavFlight/inav/files/2318202/dump.txt)

Could you please point me in the right direction? Thanks for your time!

RCnator, 2018-08-26:
I also get an error in "settings validated" besides "navigation is safe" even though I have 17 sats. My "report_cell_voltage" is set to "OFF" btw. I hope my dump below is readable.

[dump.txt](https://github.com/iNavFlight/inav/files/2321320/dump.txt)

sebmirasol, 2018-08-26:
Yeah, that fixed it for me! Thanks!

closed 2018-08-26 by digitalentity

shellixyz, 2018-11-14:
@Taka1996108 Try `set report_cell_voltage = OFF`. But that's not the right way to fix an invalid setting report. You should restore the defaults and start the configuration again. If one setting is reported as invalid it means you flashed the firmware without the "erase all" option and the settings migration failed. More settings could have an extreme value and cause unexpected behaviour.

Taka1996108, 2018-11-22:
Thanks you so much!

Regards
Taka