iNavFlight/inav #3792: Blackbox doesn’t record

spacyje, 2018-08-24:
Hello,
I don't understand why my blackbox doesn't record anything.  When I get this screen on the configurator normally it's okay, right?
[image]

- feature blackbox enabled
- device sdcard 
- switch mode blackbox on
- sdcard SDHC CLASS6 8Go

I only have the freespa.ce file that is empty.

teckel12, 2018-08-24:
@spacyje My configurator looks different (shows space used and free on SD card).  Also, the "Filesystem starting..." message sounds like it's trying to access the card but it can't.

Try a different SD card or reformat the card as FAT32 with the [SD Memory Card Formatter](https://www.sdcard.org/downloads/formatter_4/).

davidngrc, 2018-08-26:
@spacyje just finish a flight in iNav 2.0 official, the blackbox works fine, I am using a 32G class 10 SD Card (some unknown chinese brand)

@teckel12 I use the iNav log viewer in https://github.com/iNavFlight/blackbox-log-viewer
I could not get the gps coordinates, nor the gps time display in the viewer.
Could you?
I remember @fiam modify some gps related info in the blackbox on https://github.com/iNavFlight/inav/issues/3531
and
https://github.com/iNavFlight/inav/issues/3516

closed 2018-11-08 by stale[bot]