iNavFlight/inav #3793: Add support for Memsic MMC5883MA magnetometer which is P2P compatible with Hmc5883

ninja-zx11, 2018-08-24:
Please add support for MMC5883MA as it performs even better than HMC5883.

Comparison done here: http://www.diva-portal.se/smash/get/diva2:1169027/FULLTEXT01.pdf

Driver already present in ardupilot Px4 firmware
: https://github.com/traxxas/PX4Firmware/commit/89fe5e7bdf7a91a5bed7a6c88f35a5b56dcbff98#diff-8989715f5461b08efdb89011d2e99b5c

[image]

digitalentity, 2018-08-25:
Are there any GPS modules with this chip?

ninja-zx11, 2018-08-25:
Small breakout boards are available from digikey which can be placed inside existing GPS puck housing I guess.Really frustrated from the bad performance of fake compass chips.As it's P2P compatible with hmc5883,I would just solder it in place of existing fake hmc5883.

digitalentity, 2018-08-27:
Ordered a sample to evaluate

ninja-zx11, 2018-08-29:
My board came today.It can be easily installed/glued inside a gps puck by removing the pin headers from the compass board.

Is it possible to have a temporary hex file with mmc5883 driver for omninxtF7 board to test this  compass board?I mean just for testing on the bench.Thanks.