iNavFlight/inav #3794: Map/Radar Mode scale position

wx4cb, 2018-08-25:
Is it possible that if you are using the map/radar mode, the scale that is in the bottom left corner can be moved or even removed as it's own toggle

For me, it takes up too much in the corner which is where I usually have my lat/long. It would be nice to be able to turn off the scale or move it to a different position

kardon18, 2018-10-10:
I agree. 
The Radar/Map function is great to use. 
But having the "Scaling" stuck in the bottom left corner, and the "Direction" stuck in the top right corner. Takes up space for other more important OSD information.
It would be useful to have the option to not including this information, and only display the Radar/Map Arrow.
Or at least have the ability to select a different location on the OSD to place the "Scaling" and "Direction".

fiam, 2019-05-07:
Fixed in #4386

PR #4386 (2019-03-15): Add dedicated OSD elements for map scale and reference
Allows moving them around (or hiding them) rather than having them
on a fixed position.

Fixes #3794
commit 7bb01678 (2019-03-15): Add dedicated OSD elements for map scale and reference

Allows moving them around (or hiding them) rather than having them
on a fixed position.

Fixes #3794
closed 2019-05-07 by fiam