iNavFlight/inav #3799: Inav 2.0 IRC Tramp 600mW setting do not work

Chefhandwerker, 2018-08-25:
600mW power setting does not work. You can choose it over the OSD. But after leaving the OSD, invalid VTX Power is displayed.
 Matek F405 Wing
 IRC Tramp Int
 Inav 2.0
 crosscheck with Betaflight Lua.
 With Betaflight Lua you can set 600mW. Sometimes with, sometimes without error in the OSD

johnlenfr, 2018-08-27:
Same thing for the Atlalt HV 25-600mW

INAV 2.0.0 / Holybro Kakute F4 V2 Flight controller / Atlalt HV 25-600mW 40CH VTX  / Tekko32 35A 4 In 1 Blheli 32 ESC

More infos on harware here :

[http://www.holybro.com/product/48](url)

[https://www.rcgroups.com/forums/showthread.php?3048970-Holybro-Kakute-F4-V2-FC-Atlalt-HV-25-600mW-40CH-VTX-Tekko32-35A-4-In-1-BLHeli-32](url)

JonnyFPV, 2018-08-29:
Same issue for me :(

BobbyCephy, 2018-09-23:
try this:
[inav_2.0.0_KAKUTEF4V2_VTX_POWER_FIX.hex.zip](https://github.com/iNavFlight/inav/files/2409019/inav_2.0.0_KAKUTEF4V2_VTX_POWER_FIX.hex.zip)
[inav_2.0.0_MATEKF405SE_VTX_POWER_FIX.hex.zip](https://github.com/iNavFlight/inav/files/2409020/inav_2.0.0_MATEKF405SE_VTX_POWER_FIX.hex.zip)

er9xuser, 2018-11-08:
Again the same problem. iNav 2.0.1 IRC Tramp settings.

MatekF405-STD IRC Tramp HV

Tramp has 5 levels but one can choose only 4. It was working in 2.0.0. rc2, rc3, rc4...why this is again broken?

teckel12, 2018-11-09:
I believe 2.0.1 was a emergency release for GPS issues. The next release will support level 5. For now, you can either compile your own or set the value to 4.

DzikuVx, 2018-11-12:
#4008

closed 2018-11-12 by DzikuVx