iNavFlight/inav #3803: OSD remaining percent based on sag compensated voltage

tiriad, 2018-08-26:
Actual Behavior

The OSD shows the remaining percentage of the battery based on the raw voltage

Desired Behavior

Be able to choose the remaining percentage  of the battery  shown on the OSD based on raw or sag compensated voltage when no other battery specification is configured  to avoid fluctuations in reading

teckel12, 2018-08-26:
@tiriad I believe there's a switch to do this. I'm fairly sure I had it turned on but didn't like it so I turned it off (as I have a current sensor which is more useful as a "fuel" indicator).

shellixyz, 2018-08-26:
It is not possible in 2.0. It will be possible in the next release by setting `bat_voltage_src` to `SAG_COMP`.

tiriad, 2018-08-26:
It is good to know that later on you will be able to activate this feature thanks for your work.

PR #3804 (2018-08-26): Calculate batt. percentage based on sag comp. voltage
If `bat_voltage_src` is set to SAG_COMP

Closes #3803
closed 2018-08-26 by tiriad