iNavFlight/inav #3806: Revisit code styling guidelines

digitalentity, 2018-08-26:
I suggest using 1TBS for coding style (variant of K&R) and writing an own (INAV-local) code style guide similar to https://www.kernel.org/doc/Documentation/process/coding-style.rst

digitalentity, 2018-08-26:
Rationale for using 1TBS style: ensure that insertion of a new line of code anywhere is always safe - an insertion will not change the execution flow of other lines.

teckel12, 2018-08-26:
@digitalentity Sorry if I caused this, I was honestly trying to adhere to the documented coding style, not start a coder controversy.

fiam, 2018-08-26:
@teckel12 you did nothing wrong. This has been discussed some months ago, but the consensus at the time was to postpone it after 2.0. Since you mentioned the style guide and we”re now past 2.0, it is a good time to take care of this.

My suggestion is to write a .clang-format file that enforces the style we’re using and run all code in src/main through it. No reason to bother with 3rd party code that it’s imported mostly verbatim. Once that has been done, add a check in Travis builds to enforce the formatting.

We should also coordinate it with BF, so exchanging code between both projects is as painless as possible. /cc @mikeller

mikeller, 2018-08-27:
@digitalentity, @fiam: 'Write down the (existent but so far unwritten) coding style for Betaflight' is a permanent top list item on my to-do list. And I totally agree that we should try and collaborate on this. Also 1TBS (never heard of that one) is aligned with what we are doing in Betaflight.

This can probably be cleaned up and used as the starting point for the Betaflight coding style:

https://github.com/betaflight/betaflight/blob/master/docs/development/CodingStyle.md

mikeller, 2018-08-27:
See betaflight/betaflight#6647.

digitalentity, 2018-08-27:
Cool! Seems we have defined a common coding style. Once it's merged to BF we can change INAV docs to adopt it as well.

mikeller, 2018-08-28:
I'll leave the pull request open for a couple of days, to give more people a chance to comment, and suggest / discuss changes.

teckel12, 2018-08-28:
@mikeller I don't mind the actual style, but the guide itself doesn't have enough examples and is far too wordy.  This could prove problematic for non-English speaking (and honestly it's a bit confusing to me as well).  There also seems to be some markdown issues with the document (pertaining to headers).

Would you mind if I took a stab at cleaning it up a bit?

teckel12, 2018-08-28:
@mikeller Actually, it was easier than I thought so I created a PR for the markdown changes and style examples: https://github.com/betaflight/betaflight/pull/6661

closed 2023-03-06 by DzikuVx