iNavFlight/inav #3809: Show waypoints in OSD's map/radar mode

nmaggioni, 2018-08-27:
## Current Behavior
The OSD's map/radar mode shows the home position relative to the craft's (or the other way around).

## Desired Behavior
The OSD map/radar mode could also show waypoints previously saved via the Mission Planner tab of the configurator, allowing the pilot to place landmarks and manually fly through them.

## Suggested Solution
Waypoints could be pulled from EEPROM at boot and handled in the OSD like the non-fixed elements are. It could be useful if they could also show up in an ordered way (Numbers? Letters?) to reflect how they were placed in the MP.

## Who does this impact? Who is this for?
All users who are on __>=v2.0.0__ and would like to manually fly missions and/or do aerial photography at specific locations which are difficult to locate by sight.

avgorbi, 2018-10-25:
+1

And mission statistics:
- current waypoint number
- distance/time to current waypoint
- direction to current waypoint

mistyk, 2019-09-03:
We have already done pretty the same feature in iNavRadar with homepoint and others uavs, so it shouldn't be to difficult to extend to waypoints.

closed 2024-02-02 by DzikuVx