iNavFlight/inav #3810: MSP query MSP_WP returns actual POS in the case of waypoint 255

YuxinPan, 2018-08-27:
## Current Behavior
MSP query MSP_WP invokes function _getWaypoint_ in _navigation.c_. And in the case of waypoint 255, it is always the current POS that is returned.

## Expected behavior
In the case of special waypoint 255, a desired coordinate for POSHOLD can be set directly from ground station, and the MSP query should reflect that rather than returning current POS. That is, it is _posControl.desiredState.pos_ that should be returned rather than _navGetCurrentActualPositionAndVelocity()->pos_.

https://github.com/iNavFlight/inav/blob/174540ad2114db33ac71ff6fad256201b472534b/src/main/navigation/navigation.c#L2383

shlapion, 2019-10-10:
Would also be interested in this feature. Are there any updates?

YuxinPan, 2020-12-22:
Special waypoint added. [Doc](https://github.com/iNavFlight/inav/wiki/MSP-Navigation-Messages) has been updated to reflect the change on special waypoint.

closed 2020-12-22 by YuxinPan