iNavFlight/inav #3811: Please can you leave me an option for 2D poshold as I use this mode all the time.

freddybloggs, 2018-08-27:
it is by far the most usefull mode for instance
1. in a field with long grass I can fly back to myself and put into poshold and lower the throttle and catch my copter, in althold/3D it tends to jump and try and hit me in the face.

2. I can stop by a tall structure and fly up and down parallel to said structure in althold/3D
it has a huge lag and can start jumping up and down.

freddybloggs, 2018-08-27:
You must live somewhere flat, we get sudden pressure changes here causing jumps, I will play around with the settings I have tried before and never found them as responsive as without althold but for the time being I have put 2D hold back into 2.0
I would rather catch it than let it land in the long grass getting grass tangled in the motors, also landing on a hard surface blows dust up also getting in the motors

teckel12, 2018-08-27:
@most of the time, jumping on altitude hold is a result of `nav_mc_hover_thr` being set incorrectly.  If you don't get a stable altitude hold, it could be the props causing it, so installing foam on the barometer can be helpful.  If you find that the barometer just isn't useful, what about just disabling the barometer?

I would argue that it's better to land in long grass or blow up some dust than cut a finger off or poke an eye out.  A simple launch pad would resolve this issue and be far more safe IMHO.

freddybloggs, 2018-08-28:
My barometer has foam both sides and is in a special printed case, and in calm weather the alt hold is perfect it's when it's blustery there are pressure blips.

teckel12, 2018-08-28:
@freddybloggs It could be more of a PID issue, see the following settings:
```
set nav_mc_pos_z_p = 50
set nav_mc_pos_z_i = 0
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
```
I believe making POSHOLD to include ALTHOLD was a good idea.  It's more understandable and simplifies controls for most.  Sure, some will miss it, but it seems more like you have a calibration issue that was causing you to need to use a now deprecated flight mode when you really shouldn't have the need.

closed 2018-11-10 by stale[bot]