iNavFlight/inav #3813: Having to set arm angle and disabling disarm kill switch for fixed wings should be on by DEFAULT

slowthisbirddown, 2018-08-27:
## Current Behavior
Accidentally disarming in air does not allow you to rearm. Potentially losing the aircraft and starting a fire.

## Steps to Reproduce
1. disarm
2. screwed

## Expected behavior
By default I should not be allowed to disarm when my throttle is above a certain threshold. Set angle should be 180 by default as well as another failsafe

## Suggested solution(s)
small angle 180 and disarm kill switch off by DEFAULT FOR FIXED WINGS

---
2.0 official

- FC Board name and vendor:
- f411 wing

digitalentity, 2019-01-06:
We don't support different defaults for different flight platforms. At the moment configuring the aircraft correctly is pilot's responsibility. Default of `disarm_kill_switch = ON` was discussed multiple times and deemed to be generally a more safe option.

Moving to Future milestone

V8Darren, 2019-06-23:
I have to say, I only fly fixed wing and totally disagree that disable_kill_switch should be on by default. The kill switch is a safety feature. It should kill the motor as soon as it’s flicked, rather than having to lower the throttle first. I know the arguments that when testing the props should be off and that all procedures should be followed. But we’re human and don’t always do everything we should, and trigger things by accident. I’d rather lose a model because I’ve disarmed in flight and couldn’t re-arm in time, than cause serious injury to anyone because I couldn’t kill the motor quick enough. 

However, if there was an extra check, I think it would be a safe feature to have. Such as

disable_kill_switch_min_alt = 1000 // range 500 - 12000

This would not disable the kill switch below 10m, therefore would be safe and still have the benefit of not being able to accidentally disarm in flight. However a system message may be useful to say that the switch has been moved.

teckel12, 2019-06-23:
@V8Darren In theory I like it, but then you don't know what to expect if it's close to the cut-off point where minimum throttle is needed to disarm. Probably better to keep it consistent as it's often in a panic-situation. Your muscle memory should then just take over.

V8Darren, 2019-06-23:
But that would be the whole point of the system message. If you accidentally hit the disarm switch in flight and you’re above the threshold, you will see a message in your goggles to alert you. Then you can correct the switch position. 

What happens at the cutoff point is simple. Above the altitude, the motor is not killed and you get a message on the OSD. Below the altitude, the motor is killed, you see the end flight screen. 

The key will be to set the default min alt so that it is higher than a person but low enough so that in normal flight disarm will not kill the motor. 3m may be a good height? It will only fail if you’re skimming the ground. 

I don’t think you’d ever learn the muscle memory to lower the throttle, then disarm. Its something you’d only do in an emergency. Whereas every time you want to disarm when flying normally, it’s 90% of the time after landing; the throttle is already lowered, so you just flick the switch to disarm. That is the most common muscle memory, just flicking the switch. 

If there’s no safety net like the min_alt, I don’t even think disable_kill_switch should be an option, it’s unsafe. If you disarm accidentally in flight and can’t re-arm, its your fault, own it. Better that than injure or cause more severe injury to someone because you can’t disarm quickly enough.

teckel12, 2019-06-23:
@V8Darren If you're flying FPV and if you have system messages turned on. And even then, in a panic-situation, reading a message may not be where your eyes or mind goes.

Imagine flying eye-of-sight (which I do on every flight at least part of the time) there's no message to tell me why it's not disarming.

I'd just prefer predictability, where I can count on the same thing in all cases and allow muscle memory to do what it does. It's automatic for me to drop the throttle and hit kill in a crash. But that's me...

V8Darren, 2019-06-24:
I still fail to see where the panic situation is. If you’re flying FPV, I’d say you’re 99% likely to be above 3m off the ground. If you flick the switch accidentally, you just see a message on the OSD saying that you’ve flicked the switch. The model keeps flying as it has done. You just think that you’re and idiot and put the switch back in the right position when you can. There is no rush or panic, just a little self-deprecation. 

If you’re flying line of sight, other than take-off and landing you will be above the minimum altitude threshold. I agree, there would be no feedback; but it wouldn’t be an issue. Of you’re on a landing approach, with a 3m trigger altitude, you would be fine with a deadstick landing. You would already be set up and would be able to land without a problem. 

Ah, and there lies the problem. Quads may well be different, when I flew mine and crashed, I just flicked the switch, didn’t care where the throttle was. That’s the safer way to do it as iNav won’t let you re-arm if the throttle is not lowered. With fixed wing, you’ll never lower the throttle to disarm, as pretty much all the time the throttle will already be low. The only exception would be if you taxi an aircraft with wheels. But even then, the brain doesn’t associate the lowering the throttle with the disarm. My natural disarm reaction is always to just flick the switch. 

I don’t think disarm_kill_switch should even be available on a quad. With 4 spinning motors and and ability to rapidly change direction, killing the motors should be instant. This isn’t about crashing this is about times when a human can be put in danger, no matter how rare those times are. The most important thing is that the motors stop immediately. Muscle memory learns that switch Sx kills the motor, so don’t flick it in flight.

teckel12, 2019-06-24:
@V8Darren  I believe our perspectives are quite different, which is a good example of why making changes like this can be beneficial to some and not to others.  For me, `disarm_kill_switch` is VERY useful.  It's a model/motor safety feature.  And most quad pilots develop the muscle memory to quickly drop the throttle and hit the arm switch at virtually the same time.  But at the same time, protect against accidently bumping the arm switch.  Again, you don't see my and most quad-flyer's perspective so you don't have the muscle memory to kill the props when you need to for safety reasons.

Agree to disagree, but `disarm_kill_switch` will never be removed for quads, that's for SURE.

DzikuVx, 2020-02-19:
INAV Configurator defaults dialog sets `small_angle` to `180` when Airplane is selected. We will not change `disarm_kill_switch` default value tho as `ON` is safer option

closed 2020-02-19 by DzikuVx