iNavFlight/inav #3814: VTx freq and power setting inav 2.0.0

1Steven, 2018-08-27:
For immediate help, just ask your question on one of the following platforms:

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)
* [IRC channel](irc://irc.freenode.net/#inavflight)
* [Slack channel](https://publicslack.com/slacks/inavflight/invites/new)

You can also read public documentations or watch video tutorials:

* [Official documentation](https://github.com/iNavFlight/inav/tree/master/docs)
* [Official Wiki](https://github.com/iNavFlight/inav/wiki)
* [Video series by Painless360](https://www.youtube.com/playlist?list=PLYsWjANuAm4qdXEGFSeUhOZ10-H8YTSnH)
* [Video series by Paweł Spychalski](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLloACrA6f1_daCjhqY2x0fB)

1Steven, 2018-08-27:
The Furious Stealth Long Range Bluetooth does not work with 2.0.0. I believe there is a conflict with the serial port and inav trying to communicate with the VTx. Also, when I enable the power select the highest power is never achieved. The Furious VTx has 25mw, 200mw and 700mw but can only go to 200mw max using inav. What I want is to either get the inav to control power correctly or disable the entire function so I can control frequency and power with the Furious bluetooth. I hope that's clear.

1Steven, 2018-08-27:
Sorry, I seem to be a cave man on using this issues feature.

JonnyFPV, 2018-08-29:
Same issue for me :(

gregm412, 2018-10-02:
I have the same issue. I'm using the f35 and furious long range vtx. It was working for me on 2rc4 but today I updated to stable 2.0 and can not get this feature to work again. If I choose power level 3 it will change to 200mw but every other power option does nothing. Please Help.

closed 2018-12-15 by stale[bot]